leifliddy/asahi-fedora-usb #2: script messes up when mounting

monomycelium, 2022-10-23:
Hello. I wish to triple-boot my M1 Mac with Arch Linux, Fedora and macOS. macOS and Arch Linux ARM works fine, but I need Fedora for virtualisation on `aarch64`. So I found this project. Kudos to your hard work!

However, the script works smoothly (on Arch Linux ARM) only until the mounting stage. Here's the thing I do not like about scripts: they don't always work. It makes debugging harder as well, and users can't even learn how the project works. But anyway, here's the output from the script:

<details>
<summary>output</summary>

```
[root@asahi asahi-fedora-usb]# ./build.sh -d /dev/sdb
### Unmounting usb partitions...
### Preparing USB device...
GPT fdisk (gdisk) version 1.0.9

Partition table scan:
  MBR: protective
  BSD: not present
  APM: not present
  GPT: present

Found valid GPT with protective MBR; using GPT.

Command (? for help): This option deletes all partitions and creates a new protective MBR.
Proceed? (Y/N): 
Command (? for help): Partition number (1-128, default 1): First sector (34-60661726, default = 2048) or {+-}size{KMGTP}: Last sector (2048-60661726, default = 60659711) or {+-}size{KMGTP}: Current type is 8300 (Linux filesystem)
Hex code or GUID (L to show codes, Enter = 8300): Changed type of partition to 'EFI system partition'

Command (? for help): Partition number (2-128, default 2): First sector (34-60661726, default = 1230848) or {+-}size{KMGTP}: Last sector (1230848-60661726, default = 60659711) or {+-}size{KMGTP}: Current type is 8300 (Linux filesystem)
Hex code or GUID (L to show codes, Enter = 8300): Changed type of partition to 'Linux filesystem'

Command (? for help): Partition number (3-128, default 3): First sector (34-60661726, default = 3328000) or {+-}size{KMGTP}: Last sector (3328000-60661726, default = 60659711) or {+-}size{KMGTP}: Current type is 8300 (Linux filesystem)
Hex code or GUID (L to show codes, Enter = 8300): Changed type of partition to 'Linux filesystem'

Command (? for help): 
Final checks complete. About to write GPT data. THIS WILL OVERWRITE EXISTING
PARTITIONS!!

Do you want to proceed? (Y/N): OK; writing new GUID partition table (GPT) to /dev/sdb.
The operation has completed successfully.
mkfs.fat 4.2 (2021-01-31)
mke2fs 1.46.5 (30-Dec-2021)
/dev/sdb2 contains a ext4 file system labelled 'fedora-usb-boot'
        last mounted on Sun Oct 23 11:26:53 2022
Creating filesystem with 262144 4k blocks and 65536 inodes
Filesystem UUID: a1492762-3fe2-4908-a8b9-118439becd26
Superblock backups stored on blocks: 
        32768, 98304, 163840, 229376

Allocating group tables: done                            
Writing inode tables: done                            
Creating journal (8192 blocks): done
Writing superblocks and filesystem accounting information: done

mke2fs 1.46.5 (30-Dec-2021)
/dev/sdb3 contains a ext4 file system labelled 'fedora-usb-root'
        last mounted on Sun Oct 23 11:26:39 2022
Creating filesystem with 7166464 4k blocks and 1794048 inodes
Filesystem UUID: d747cb2a-aff1-4e47-8a33-c4d9b7475df9
Superblock backups stored on blocks: 
        32768, 98304, 163840, 229376, 294912, 819200, 884736, 1605632, 2654208, 
        4096000

Allocating group tables: done                            
Writing inode tables: done                            
Creating journal (32768 blocks): done
Writing superblocks and filesystem accounting information: done   

egrep: warning: egrep is obsolescent; using grep -E
‣ Removing output files…
--2022-10-23 11:28:22--  https://leifliddy.com/asahi-linux/asahi-linux.repo
Loaded CA certificate '/etc/ssl/certs/ca-certificates.crt'
Resolving leifliddy.com (leifliddy.com)... 46.101.247.100
Connecting to leifliddy.com (leifliddy.com)|46.101.247.100|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 536 [text/plain]
Saving to: ‘mkosi.skeleton/etc/yum.repos.d/asahi-linux.repo’

mkosi.skeleton/etc/yum.repo 100%[=========================================>]     536  --.-KB/s    in 0s      

2022-10-23 11:28:23 (10.3 MB/s) - ‘mkosi.skeleton/etc/yum.repos.d/asahi-linux.repo’ saved [536/536]

--2022-10-23 11:28:23--  https://leifliddy.com/.vendorfw/all_firmware.tar.gz
Loaded CA certificate '/etc/ssl/certs/ca-certificates.crt'
Resolving leifliddy.com (leifliddy.com)... 46.101.247.100
Connecting to leifliddy.com (leifliddy.com)|46.101.247.100|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 9925835 (9.5M) [text/plain]
Saving to: ‘mkosi.skeleton/boot/efi/asahi/all_firmware.tar.gz’

mkosi.skeleton/boot/efi/asa 100%[=========================================>]   9.47M  4.65MB/s    in 2.0s    

2022-10-23 11:28:26 (4.65 MB/s) - ‘mkosi.skeleton/boot/efi/asahi/all_firmware.tar.gz’ saved [9925835/9925835]

--2022-10-23 11:28:26--  https://leifliddy.com/.vendorfw/firmware.tar
Loaded CA certificate '/etc/ssl/certs/ca-certificates.crt'
Resolving leifliddy.com (leifliddy.com)... 46.101.247.100
Connecting to leifliddy.com (leifliddy.com)|46.101.247.100|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 10772480 (10M) [text/plain]
Saving to: ‘mkosi.skeleton/boot/efi/vendorfw/firmware.tar’

mkosi.skeleton/boot/efi/ven 100%[=========================================>]  10.27M  5.12MB/s    in 2.0s    

2022-10-23 11:28:28 (5.12 MB/s) - ‘mkosi.skeleton/boot/efi/vendorfw/firmware.tar’ saved [10772480/10772480]

--2022-10-23 11:28:28--  https://leifliddy.com/.vendorfw/manifest.txt
Loaded CA certificate '/etc/ssl/certs/ca-certificates.crt'
Resolving leifliddy.com (leifliddy.com)... 46.101.247.100
Connecting to leifliddy.com (leifliddy.com)|46.101.247.100|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 15169 (15K) [text/plain]
Saving to: ‘mkosi.skeleton/boot/efi/vendorfw/manifest.txt’

mkosi.skeleton/boot/efi/ven 100%[=========================================>]  14.81K  --.-KB/s    in 0s      

2022-10-23 11:28:29 (166 MB/s) - ‘mkosi.skeleton/boot/efi/vendorfw/manifest.txt’ saved [15169/15169]

‣ Detaching namespace
‣ Setting up package cache…
‣ Setting up package cache /home/ay/Documents/cache/asahi-fedora-usb/.mkosi-cppt9h20 complete
‣ Setting up temporary workspace.
‣ Temporary workspace set up in /home/ay/Documents/cache/asahi-fedora-usb/.mkosi-okw7h72j
‣ Running second (final) stage…
‣  Mounting image…
‣  Setting up basic OS tree…
‣  Copying in skeleton file trees…
‣  Mounting Package Cache
‣  Installing Fedora Linux…
‣   Mounting API VFS
‣ Error: yum not found in PATH.
‣   (Unmounting API VFS)
‣  (Unmounting Package Cache)
‣  (Unmounting image)
### Cleaning up...
### Mounting usb partitions and copying files...
rsync: [sender] change_dir "/home/ay/Documents/cache/asahi-fedora-usb/mkosi.rootfs" failed: No such file or directory (2)
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1336) [sender=3.2.7]
mount: mnt_usb/boot: mount point does not exist.
       dmesg(1) may have more information after failed mount system call.
rsync: [sender] change_dir "/home/ay/Documents/cache/asahi-fedora-usb/mkosi.rootfs/boot" failed: No such file or directory (2)
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1336) [sender=3.2.7]
mount: mnt_usb/boot/efi: mount point does not exist.
       dmesg(1) may have more information after failed mount system call.
rsync: [sender] change_dir "/home/ay/Documents/cache/asahi-fedora-usb/mkosi.rootfs/boot/efi" failed: No such file or directory (2)
rsync: [Receiver] change_dir#3 "/home/ay/Documents/cache/asahi-fedora-usb/mnt_usb/boot" failed: No such file or directory (2)
rsync error: errors selecting input/output files, dirs (code 3) at main.c(827) [Receiver=3.2.7]
### Setting uuids for partitions in /etc/fstab...
sed: can't read mnt_usb/etc/fstab: No such file or directory
sed: can't read mnt_usb/etc/fstab: No such file or directory
sed: can't read mnt_usb/etc/fstab: No such file or directory
### Running systemd-machine-id-setup...
chroot: failed to run command ‘systemd-machine-id-setup’: No such file or directory
chroot: failed to run command ‘echo’: No such file or directory
### Updating GRUB...
mount: mnt_usb/proc: mount point does not exist.
       dmesg(1) may have more information after failed mount system call.
==> ERROR: failed to setup chroot mnt_usb
### Creating BLS (/boot/loader/entries/) entry...
chroot: failed to run command ‘/image.creation/create.bls.entry’: No such file or directory
### Enabling system services...
chroot: failed to run command ‘systemctl’: No such file or directory
### Disabling systemd-firstboot...
chroot: failed to run command ‘rm’: No such file or directory
find: ‘mnt_usb/boot/efi/’: No such file or directory
chmod: missing operand after ‘700’
Try 'chmod --help' for more information.
### Unmounting usb partitions...
umount: mnt_usb/boot/efi: no mount point specified.
umount: mnt_usb/boot: no mount point specified.
### Done
```

</details>

Hope you can help!

leifliddy, 2022-10-23:
It's the `mkosi` part that's not working. 
The output should look like this: 

```
--2022-10-23 07:50:06--  https://leifliddy.com/.vendorfw/manifest.txt
Resolving leifliddy.com (leifliddy.com)... 46.101.247.100
Connecting to leifliddy.com (leifliddy.com)|46.101.247.100|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 15169 (15K) [text/plain]
Saving to: ‘mkosi.skeleton/boot/efi/vendorfw/manifest.txt’

mkosi.skeleton/boot/efi/vendo 100%[===============================================>]  14.81K  --.-KB/s    in 0.001s  

2022-10-23 07:50:06 (15.8 MB/s) - ‘mkosi.skeleton/boot/efi/vendorfw/manifest.txt’ saved [15169/15169]

‣ Detaching namespace
‣ Setting up package cache…
‣ Setting up package cache /root/asahi-fedora-usb/.mkosi-v66bkrrx complete
‣ Setting up temporary workspace.
‣ Temporary workspace set up in /root/asahi-fedora-usb/.mkosi-0vqt7i82
‣ Running second (final) stage…
‣  Mounting image…
‣  Setting up basic OS tree…
‣  Copying in skeleton file trees…
‣  Mounting Package Cache
‣  Installing Fedora Linux…
‣   Mounting API VFS
Asahi Fedora 36 - aarch64                                                             1.6 MB/s | 570 kB     00:00    
Fedora 36 - aarch64                                                                   9.7 MB/s |  77 MB     00:07    
Fedora 36 - aarch64 - Updates                                                         8.4 MB/s |  27 MB     00:03    
Dependencies resolved.
======================================================================================================================
 Package                               Architecture    Version                              Repository           Size
======================================================================================================================
Installing:
 arch-install-scripts                  noarch          24-3.fc36                            fedora               27 k
 asahi-fwextract                       noarch          0.5pre1-2.fc36                       asahi-linux          40 k
 asahi-scripts                         noarch          20220919-1.fc36                      asahi-linux          13 k
 audit                                 aarch64         3.0.9-1.fc36                         updates             253 k
 bash-completion                       noarch          1:2.11-6.fc36                        fedora              291 k
 binutils                              aarch64         2.37-36.fc36                         updates             6.4 M
 btrfs-progs                           aarch64         6.0-1.fc36                           updates             971 k
 cloud-utils-growpart                  noarch          0.31-10.fc36                         fedora               30 k
 dhcp-client                           aarch64         12:4.4.3-4.P1.fc36                   updates             777 k
 diffutils                             aarch64         3.8-2.fc36                           fedora              368 k
```
And judging from your output: ```Error: yum not found in PATH.```   
It's because you don't have `dnf` (or `yum`) installed.    
I'm not sure why it's doing a check for yum, `dnf` is what it should be using. 

When you install `mkosi` on Fedora, the following packages get installed as dependencies.

```
 archlinux-keyring            noarch 20220927-1.fc37      updates-testing 1.1 M
 dpkg                         x86_64 1.21.9-1.fc37        fedora          1.5 M
 pacman                       x86_64 6.0.1-5.fc37         fedora          750 k
 pacman-filesystem            noarch 6.0.1-5.fc37         fedora          7.8 k
 ...
 debian-keyring               noarch 2021.1.1-4.fc37      fedora           94 k
 debootstrap                  noarch 1.0.127-2.fc37       fedora           86 k
```

These packages allow `mkosi` to build Debian or Arch images on Fedora. 
So I'm guessing that Arch isn't installing `dnf` as a dependency of `mkosi` 
As I pointed out in the `readme`

```Note: I've only built this on a Fedora system, so YMMV on other distros.```
I'm not an Arch or Debian user. So if you're running this script on `Arch`, there might be steps that are either missing, or it may not work at all. I have no idea. 

In any case,m I just added `set -e` to the top of the `build.sh` script. So the script will now error out and exit upon encountering the first failure. I would try and containerize this, but that would conflict with `mkosi` on some level. I would need to test that out. 
So the easiest way to get this to work is to build it on a Fedora system.   
You can build this on a `Fedora x86_64` VM as long as you have the `qemu-user-static` package installed.

monomycelium, 2022-10-23:
thank you for that thorough explanation! i'll try out the two solutions soon.

monomycelium, 2022-10-23:
I'm using a Fedora virtual machine (with KVM), but running the script only gives me this:

```
[root@fedora asahi-fedora-usb]# ./build.sh -d /dev/sda
### Unmounting usb partitions...
```

leifliddy, 2022-10-23:
Can you post the following output?
```df -h | grep sda```

leifliddy, 2022-10-23:
If you're still having issues, modify the top of the script to this: 
```
#!/bin/bash

set -e
set -x
```
That `-x` will show exactly what the script is trying to do. 
I'd be helpful if you could post the output after setting that.

leifliddy, 2022-10-23:
My server logs are showing that you ran that a few times 
```
101.x.x.x - - [23/Oct/2022:11:46:25 +0200] "GET /asahi-linux/asahi-linux.repo HTTP/1.1" 200 536 "-" "Wget/1.21.3"
101.x.x.x - - [23/Oct/2022:11:46:28 +0200] "GET /.vendorfw/all_firmware.tar.gz HTTP/1.1" 200 9925835 "-" "Wget/1.21.3"
101.x.x.x - - [23/Oct/2022:11:46:31 +0200] "GET /.vendorfw/firmware.tar HTTP/1.1" 200 10772480 "-" "Wget/1.21.3"
101.x.x.x - - [23/Oct/2022:11:46:32 +0200] "GET /.vendorfw/manifest.txt HTTP/1.1" 200 15169 "-" "Wget/1.21.3"
101.x.x.x - - [23/Oct/2022:11:46:33 +0200] "GET /asahi-linux/36/aarch64/repodata/repomd.xml HTTP/1.1" 200 3098 "-" "libdnf"
101.x.x.x - - [23/Oct/2022:11:46:34 +0200] "GET /asahi-linux/36/aarch64/repodata/a773766b1c60c19b4a8a16e25cac071373a4c0b9ce144cb7d41744eef3f9fc3c-primary.xml.gz HTTP/1.1" 200 183032 "-" "libdnf"
101.x.x.x - - [23/Oct/2022:11:46:35 +0200] "GET /asahi-linux/36/aarch64/repodata/811d6006d15382f7553a8772cb0fd8f394eca94bb1aa06a6b0bf138c0067aa1a-filelists.xml.gz HTTP/1.1" 200 400597 "-" "libdnf"
101.x.x.x - - [23/Oct/2022:11:47:10 +0200] "GET /asahi-linux/36/aarch64/asahi-fwextract-0.5pre1-2.fc36.noarch.rpm HTTP/1.1" 200 41291 "-" "libdnf"
101.x.x.x - - [23/Oct/2022:11:47:10 +0200] "GET /asahi-linux/36/aarch64/asahi-scripts-20220919-1.fc36.noarch.rpm HTTP/1.1" 200 13480 "-" "libdnf"
101.x.x.x - - [23/Oct/2022:11:47:10 +0200] "GET /asahi-linux/36/aarch64/kernel-6.1.0-0.rc1.asahi2.1.fc36.aarch64.rpm HTTP/1.1" 200 6293 "-" "libdnf"
101.x.x.x - - [23/Oct/2022:11:47:11 +0200] "GET /asahi-linux/36/aarch64/lzfse-devel-1.0-1.fc36.aarch64.rpm HTTP/1.1" 200 8858 "-" "libdnf"
101.x.x.x - - [23/Oct/2022:11:47:11 +0200] "GET /asahi-linux/36/aarch64/lzfse-libs-1.0-1.fc36.aarch64.rpm HTTP/1.1" 200 25615 "-" "libdnf"
101.x.x.x - - [23/Oct/2022:11:47:12 +0200] "GET /asahi-linux/36/aarch64/kernel-modules-6.1.0-0.rc1.asahi2.1.fc36.aarch64.rpm HTTP/1.1" 200 8681861 "-" "libdnf"
101.x.x.x - - [23/Oct/2022:11:47:13 +0200] "GET /asahi-linux/36/aarch64/kernel-core-6.1.0-0.rc1.asahi2.1.fc36.aarch64.rpm HTTP/1.1" 200 17580053 "-" "libdnf"
101.x.x.x - - [23/Oct/2022:11:48:03 +0200] "GET /asahi-linux/RPM-GPG-KEY-asahi HTTP/1.1" 200 648 "-" "libdnf"
```
What happened after you ran that? That looks like a successful install.

monomycelium, 2022-10-23:
Hi there again. Interesting that you can actually see that. That indeed seems to be my IP address. Sorry for hogging your server.

Anyway, the disks are not mounted, so you can't see anything with `df -h | grep sda`. I'm not going to mount them again (because I'm too embarrassed to say that I have actually quit the script in the middle of making a filesystem, which I'll fix later) but the disks were empty after the build script, except for `lost+found`, which I'll assume was empty. Take a look at my partitions, though:

```
Device       Start      End  Sectors  Size Type
/dev/sda1     2048  1230847  1228800  600M EFI System
/dev/sda2  1230848  3327999  2097152    1G Linux filesystem
/dev/sda3  3328000 60659711 57331712 27.3G Linux filesystem
```

They're pretty fine. But anyway. Could we get documentation to prepare this instead? I hate having to rely on scripts. I get that this project is not backed up by too many people, but I could use a brief guide.

leifliddy, 2022-10-23:
Ok I see the issue

```
./build.sh: line 120: rsync: command not found
mount: mnt_usb/boot: mount point does not exist.
       dmesg(1) may have more information after failed mount system call.
./build.sh: line 122: rsync: command not found
mount: mnt_usb/boot/efi: mount point does not exist.
       dmesg(1) may have more information after failed mount system call.
./build.sh: line 124: rsync: command not found
```

Did you pull in the latest version of the script? If you had that `set -e` the script, then it would have failed (and exited) at `rsync` vs trying to mount and chroot into a non-existent drive. 
Please use the latest version, I put that out there for you. 

Ok, it's failing because you don't have `rsync` installed. I just updated the readme to include that package. You can install it with `dnf install rsync`

monomycelium, 2022-10-23:
Oh shoot, I did not see that. Sorry. Anyway, I would have used the updated script, but it does not work:

> ```
> [root@fedora asahi-fedora-usb]# ./build.sh -d /dev/sda
> ### Unmounting usb partitions...
> ```
> 
> In fact, running this on Arch after pulling the new commit gives me the same thing. To revert back, I just checked git out to `8d84186`.

leifliddy, 2022-10-23:
Can you please do the following. 
1.  ensure `mount | grep mnt_usb` does not return anything. 
2. wipe the drive with `wipefs -a /dev/sda`
3. re-run the script again, and post the full output

monomycelium, 2022-10-23:
I did not read that in time, sorry, and I'm at step three right now. It's bootstrapping `dnf` stuff.

leifliddy, 2022-10-23:
Ok no worries. Let me know how it goes. I think I know why it was getting stuck at 
```### Unmounting usb partitions...``` and that's because the partitions were empty. 
I can modify the script to check for that.

monomycelium, 2022-10-23:
I think the script worked! Let me try booting on my Mac.

monomycelium, 2022-10-23:
The script worked. Thank you for staying by my side while I was trying to get this to work!

On my Mac, I just stopped autoboot and entered `run bootcmd_usb0` as mentioned in the readme. Kernel worked, and `systemd` started.

But `systemd` got stuck what starting some service to control the backlight. Next thing I know, my Mac restarts and autoboots into Arch Linux.

leifliddy, 2022-10-23:
Oh really!? What macbook model do you have?

monomycelium, 2022-10-23:
2020 M1 MacBook Air.

leifliddy, 2022-10-23:
That's odd I have the exact same model macbook.   
I just built a new usb drive -- and it booted fine. Could you try booting it again?

leifliddy, 2022-10-23:
Hmmm...do you know what firmware version you're currently on?

monomycelium, 2022-10-23:
Firmware version as in `m1n1` or something?

monomycelium, 2022-10-23:
> That's odd I have the exact same model macbook.
> I just built a new usb drive -- and it booted fine. Could you try booting it again?
> (I've never run into that issue before)

same thing: it hangs here and reboots

[image]

leifliddy, 2022-10-23:
One sec...

leifliddy, 2022-10-23:
I think I might know what the issue is but I'm not sure. If you boot into macos then goto --> "about this mac". What version does it show? It should be 12.x 
I'm running 12.5. Just wondering if you're on a different version. If so we might need to copy over the firmware archive from Arch --> Fedora and see if that works.

monomycelium, 2022-10-23:
I'm actually running macOS 12.6.

monomycelium, 2022-10-23:
Oh, and I happen to have a question: is the USB environment a typical live environment or a normal installation of Fedora? Like, can I use it like I would use a typical minimal Fedora installation?

leifliddy, 2022-10-23:
Damn, ok let me upgrade my system to `12.6` and see if the firmware here changes:
```
[root@fedora ~]# ll /boot/efi/asahi/all_firmware.tar.gz 
-rwx------ 1 root root 14086524 Aug 21 04:21 /boot/efi/asahi/all_firmware.tar.gz
```
Actually, I need to make some a couple modifications to that `build.sh` script to account for the new firmware format. We no longer need to pull in these two files 
```
[root@fedora ~]# ll /boot/efi/vendorfw/
total 15892
-rwx------ 1 root root 16250880 Jul 14 02:00 firmware.tar
-rwx------ 1 root root    18525 Jul 14 02:00 manifest.txt
```
As these files are now extracted from `all_firmware.tar.gz`
I just need a bit of time of upgrade my system and test this out.

monomycelium, 2022-10-23:
Sure. I can wait.

leifliddy, 2022-10-23:
It's an actual installation of Fedora. But, It wouldn't boot on non-apple silicon devices as it only has device tree support for Apple silicon.

monomycelium, 2022-10-23:
I love your commitment to this project! Thanks!

leifliddy, 2022-10-23:
That sounds pretty cool!
Ok, I've made a number of changes to this project.
1. We no longer need to pull in the firmware externally. The firmware will be located and extracted from the internal nvme drive.
2. I sorted out a minor issue with the first-boot service so that the firmware will be extracted before the udev service runs (so the wifi interface will now come up on the first boot) 
3. I sorted out why the script was occasionally handing on that `### Unmounting usb partitions...` function
4. I created a wipe function so that if you've previously installed Fedora to the usb drive, you can reinstall Fedora without having to reformat the drive by providing the `-w` argument  ie 
`./build.sh -wd /dev/sda`

However, I don't think any of these things are going to help you with your issue. Even if the firmware wasn't extracted at all, the system will still boot. I'm beginning to think it might be `m1n1` related. All of my packages are based on the bleeding edge Arch packages found here: https://de.mirror.asahilinux.org/aarch64/asahi-dev/
Can you enable this repo in Arch and update your packages with this development repo (or at the very least, update the `m1n1` package from this repo). The versions of these packages have been working fine for me.   
Hopefully, when you update `m1n1` and the `update-m1n1` script updates `/boot/efi/m1n1/boot.bin` -- the usb drive "should" boot properly after that. If it doesn't, let me know.

monomycelium, 2022-10-24:
Thanks! I'll try that out.

monomycelium, 2022-10-24:
thank you for that thorough explanation! i'll try out the two solutions soon.

it's working!!! i'll let you know more later.

closed 2022-10-25 by leifliddy