leifliddy/asahi-fedora-usb #3: Build Error

isteiger, 2023-01-02:
Hello! I'm trying to get this to work inside of a Fedora 37 VM in Parallels on my M1 MBP. It seems to get pretty far in, but then seems to fail at `fallocate: fallocate failed: keep size mode is unsupported`. I really don't know much of what I'm doing here as I haven't dealt much with OS builds, but I did install all the prerequisite packages. Attached is the full log:

[pastebin](https://pastebin.com/uipP5VZx)

leifliddy, 2023-01-02:
I just ran this script on my macbook m1 running fedora, that last part should look like this 

```‣    Installing boot loader…
Copied "/usr/lib/systemd/boot/efi/systemd-bootaa64.efi" to "/efi/EFI/systemd/systemd-bootaa64.efi".
Copied "/usr/lib/systemd/boot/efi/systemd-bootaa64.efi" to "/efi/EFI/BOOT/BOOTAA64.EFI".
Random seed file /efi/loader/random-seed successfully written (32 bytes).
‣    Recording packages in manifest…
‣    Resetting machine ID
‣    Removing random seed
‣    Trimming File System
/var/tmp/mkosi-b69fkx7c/root: 499.5 MiB (523722752 bytes) trimmed
‣   Unmounting image…
‣   Mounting image…
‣    Generating combined kernel + initrd boot file…
‣   Unmounting image…
‣   Mounting image…
‣   Unmounting image…
‣   Mounting image…
‣   Unmounting image…
‣  Detaching /dev/loop0
‣ Digging holes into output file /root/asahi-fedora-usb/.mkosi-ppokxrd6…
‣ Linking image file…
‣  Changing ownership of output file fedora.raw to user root…
‣  Changed ownership of fedora.raw
‣ Linked fedora.raw
‣ Linking split kernel image…
‣  Changing ownership of output file fedora.vmlinuz to user root…
‣  Changed ownership of fedora.vmlinuz
‣ Linked fedora.vmlinuz
‣ Linking split initrd…
‣  Changing ownership of output file fedora.initrd to user root…
‣  Changed ownership of fedora.initrd
‣ Linked fedora.initrd
‣ Linking split cmdline…
‣  Changing ownership of output file fedora.cmdline to user root…
‣  Changed ownership of fedora.cmdline
‣ Linked fedora.cmdline
‣ Saving manifest fedora.raw.manifest
‣  Changing ownership of output file fedora.raw.manifest to user root…
‣  Changed ownership of fedora.raw.manifest
‣ Resulting image size is 1.5G, consumes 770.3M.
...
```
So your build is failing on that `fallocate` command that `mkosi` uses to create the raw image with 
```
fallocate: fallocate failed: keep size mode is unsupported
...
run(["fallocate", "--dig-holes", data.name])
```
I have no idea why that would fail. I'm not a mac user and I've never used parallels -- so I'm not sure what sort of  limitations it has. It shouldn't be too hard to sort out though. 
Are you building this image on the local filesystem within the VM?   
Or are you building this on a mounted CIFS/SMB or NFS share?   
Or a directory that's shared between the VM and the host?
**you should be building this on the local filesystem -- not on a shared volume 

Also, what filesystem is the VM using? Are you using logical volumes?

Can you try running the following commands from within that `/media/psf/Home/Downloads/` directory?

```
truncate -s 100M root.img
mkfs.ext4 root.img 
mount root.img /mnt

# copy some random files to /mnt (just to put some data on it)

# ensure the filesystem is (loop) mounted
[root@fedora-vm ~]# df -h
/dev/loop0       89M   67M   15M  82% /mnt

fstrim /mnt
umount /mnt

fallocate --dig-holes root.img
```

Let me know if either the `fstrim` or `fallocate` commands error out.

leifliddy, 2023-01-02:
> I really don't know much of what I'm doing here

You and me both ; )

isteiger, 2023-01-02:
Right, so the way I was doing it was the Parallels mounted share, which would be a network share. That is likely the issue. 

After doing what you suggested, [this is the output](https://pastebin.com/xi8Fk1X5)

`fstrim` seems to error out with `fstrim: mnt/: the discard operation is not supported` while `fallocate` still errors out with `fallocate: fallocate failed: keep size mode is unsupported`

Trying again, but moving to a local directory seems to have fixed the issue. I'm not entirely sure why it would fail on a mounted volume because there should be no difference. If you want to continue trying to solve the issue let me know, but probably the best thing to do is just put something the readme saying to make sure you are doing it on a local directory.

leifliddy, 2023-01-02:
`fstrim` is failing because `root.img` is located on that shared directory. 

So it says here:
http://download.parallels.com/doc/pcs/html/Parallels_Cloud_Server_Users_Guide/35694.htm

`Shared folders are not a POSIX-compliant file system and are designed just for sharing files and folders`

And that's really the key here. It's not whether the directory is local or not (per se)-- but whether the underlying file system is `POSIX-compliant`

Anyways, I'm glad you got it working!

closed 2023-01-02 by leifliddy

isteiger, 2023-01-02:
Ah ok, that clears it up. Thanks for your help!