leifliddy/asahi-fedora-usb #4: Another build failure on x86_64 Fedora 37 (objcopy failing?)

felixphew, 2023-01-18:
Hi! After installing the listed dependencies on my (x86_64) Fedora 37 box, I encountered an error when running the latest (as of this morning) install script.

<details><summary>log</summary>

```
### Preparing USB device...
GPT fdisk (gdisk) version 1.0.9

Partition table scan:
  MBR: protective
  BSD: not present
  APM: not present
  GPT: present

Found valid GPT with protective MBR; using GPT.

Command (? for help): This option deletes all partitions and creates a new protective MBR.
Proceed? (Y/N): 
Command (? for help): Partition number (1-128, default 1): First sector (34-30285790, default = 2048) or {+-}size{KMGTP}: Last sector (2048-30285790, default = 30283775) or {+-}size{KMGTP}: Current type is 8300 (Linux filesystem)
Hex code or GUID (L to show codes, Enter = 8300): Changed type of partition to 'EFI system partition'

Command (? for help): Partition number (2-128, default 2): First sector (34-30285790, default = 4196352) or {+-}size{KMGTP}: Last sector (4196352-30285790, default = 30283775) or {+-}size{KMGTP}: Current type is 8300 (Linux filesystem)
Hex code or GUID (L to show codes, Enter = 8300): Changed type of partition to 'Linux filesystem'

Command (? for help): 
Final checks complete. About to write GPT data. THIS WILL OVERWRITE EXISTING
PARTITIONS!!

Do you want to proceed? (Y/N): OK; writing new GUID partition table (GPT) to /dev/sdb.
The operation has completed successfully.
mkfs.fat 4.2 (2021-01-31)
mke2fs 1.46.5 (30-Dec-2021)
Creating filesystem with 3260928 4k blocks and 816000 inodes
Filesystem UUID: d747cb2a-aff1-4e47-8a33-c4d9b7475df9
Superblock backups stored on blocks: 
	32768, 98304, 163840, 229376, 294912, 819200, 884736, 1605632, 2654208

Allocating group tables: done                            
Writing inode tables: done                            
Creating journal (16384 blocks): done
Writing superblocks and filesystem accounting information: done   

‣ Removing output files…
--2023-01-18 11:36:33--  https://leifliddy.com/asahi-linux/asahi-linux.repo
Resolving leifliddy.com (leifliddy.com)... 46.101.247.100
Connecting to leifliddy.com (leifliddy.com)|46.101.247.100|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 536 [text/plain]
Saving to: ‘mkosi.skeleton/etc/yum.repos.d/asahi-linux.repo’

mkosi.skeleton/etc/yum.repos.d/asahi-linux.repo                     100%[===================================================================================================================================================================>]     536  --.-KB/s    in 0s      

2023-01-18 11:36:34 (2.78 MB/s) - ‘mkosi.skeleton/etc/yum.repos.d/asahi-linux.repo’ saved [536/536]

‣ Removing output files…
‣ Detaching namespace
‣ Setting up temporary workspace.
‣ Temporary workspace set up in /var/tmp/mkosi-tk3znktv
‣ Running second (final) stage…
‣  Creating image with partition table…
Disk /home/felix/Documents/Projects/fedora-asahi/asahi-fedora-usb/.mkosi-wggusxhe: 1.45 GiB, 1556967424 bytes, 3040952 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes

>>> Script header accepted.
>>> Script header accepted.
>>> Script header accepted.
>>> Created a new GPT disklabel (GUID: 4ACF3ED1-81AA-1F44-918D-8300557DC50C).
/home/felix/Documents/Projects/fedora-asahi/asahi-fedora-usb/.mkosi-wggusxhe1: Created a new partition 1 of type 'EFI System' and of size 256 MiB.
[... 141 lines cut ...]
  File "/usr/lib/python3.11/site-packages/mkosi/__init__.py", line 8254, in run_verb
    manifest = build_stuff(config)
               ^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/mkosi/__init__.py", line 7559, in build_stuff
    image = build_image(state, manifest=manifest)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/mkosi/__init__.py", line 7342, in build_image
    install_unified_kernel(state, root_hash, mount)
  File "/usr/lib/python3.11/site-packages/mkosi/__init__.py", line 4080, in install_unified_kernel
    run(cmd)
  File "/usr/lib/python3.11/site-packages/mkosi/backend.py", line 860, in run
    return subprocess.run(cmdline, check=check, stdout=stdout, stderr=stderr, env={**os.environ, **env}, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib64/python3.11/subprocess.py", line 569, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['objcopy', '--add-section', '.osrel=/var/tmp/mkosi-tk3znktv/root/usr/lib/os-release', '--change-section-vma', '.osrel=0x20000', '--add-section', '.cmdline=/var/tmp/mkosi-tk3znktv/cmdline', '--change-section-vma', '.cmdline=0x30000', '--add-section', '.linux=/var/tmp/mkosi-tk3znktv/root/lib/modules/6.2.0-0.rc3.asahi5.1.fc37.aarch64/vmlinuz', '--change-section-vma', '.linux=0x2000000', '--add-section', '.initrd=/var/tmp/mkosi-tk3znktv/root/efi/46d1386bbc7c40baacedc67d2cbab734/6.2.0-0.rc3.asahi5.1.fc37.aarch64/initrd', '--change-section-vma', '.initrd=0x3000000', '/var/tmp/mkosi-tk3znktv/root/lib/systemd/boot/efi/linuxaa64.efi.stub', '/var/tmp/mkosi-tk3znktv/root/efi/EFI/Linux/fedora-6.2.0-0.rc3.asahi5.1.fc37.aarch64.efi']' returned non-zero exit status 1.
```

</details>

```objcopy: /var/tmp/mkosi-tk3znktv/root/lib/systemd/boot/efi/linuxaa64.efi.stub: file format not recognized``` appears to be the problem.

I can't say I'm 100% familiar with how this process works, but - maybe it's somehow calling the wrong (i.e. the host's) objcopy? Or maybe the file in question doesn't exist? (idk, the script cleans up after itself so I couldn't check)

Any advice would be appreciated. Happy to try things out to troubleshoot, of course.

leifliddy, 2023-01-18:
It seems like `mkosi` is a bit of a moving target, as they're constantly pushing out major changes.  
So crossarch building use to work -- there's an active issue for that here: https://github.com/systemd/mkosi/issues/1155
For now, this needs to be built on an `aarch64` system.

leifliddy, 2023-01-18:
I have a solution one sec....

felixphew, 2023-01-18:
I wonder how much extra work it'd be to cut `mkosi` out of the loop entirely and simply use `dnf --installroot` directly? (Not as a quick fix, of course, rather as a path to move forward if `mkosi` is unstable.)

leifliddy, 2023-01-18:
>dnf --installroot directly

That's not a bad idea. I'm not really using much of `mkosi` at this point. 

You can crossbuild this branch -- which uses grub
https://github.com/leifliddy/asahi-fedora-usb/tree/v1

It's a bit slow to jump from uboot --> grub as it uses the `shim-aa64` provided `/BOOTAA64.EFI` bootloader
I need to clean up a few things up, nothing major. I'll have time tonight to do that. 

Oh, you do need to install `mkosi` from main to build this
`python3 -m pip install --user git+https://github.com/systemd/mkosi.git`

Anyways, let me know if that works for you.

felixphew, 2023-01-18:
After a silly mistake on my part (forgetting PATH doesn't always get passed through `sudo`), yes, it works :)

closed 2023-11-14 by leifliddy