leifliddy/asahi-fedora-usb #7: Generating initramfs images…

kde31, 2023-03-15:
I have executed the script under Fedora 37.
Unfortunately I get the following errors at the end and can not do anything with it.
I would like to test the USB drive on a Mac Min M2 Pro.
Can anyone help me?
Thanks already for reading.

`Fertig!
‣      Unmounting API VFS…
‣    Unmounting Package Cache
‣    Assigning hostname
‣    Setting root password
‣    Generating initramfs images…
/usr/bin/dracut: line 1054: /sys/module/firmware_class/parameters/path: No such file or directory
‣    Installing boot loader…
Copied "/usr/lib/systemd/boot/efi/systemd-bootaa64.efi" to "/efi/EFI/systemd/systemd-bootaa64.efi".
Copied "/usr/lib/systemd/boot/efi/systemd-bootaa64.efi" to "/efi/EFI/BOOT/BOOTAA64.EFI".
Random seed file /efi/loader/random-seed successfully written (32 bytes).
‣    Recording packages in manifest…
‣    Resetting machine ID
‣    Removing random seed
‣    Trimming File System
/var/tmp/mkosi-amxlrk94/root: 486,9 MiB (510517248 Bytes) getrimmt
‣   Unmounting image…
‣   Mounting image…
‣    Generating combined kernel + initrd boot file…
objcopy: /var/tmp/mkosi-amxlrk94/root/lib/systemd/boot/efi/linuxaa64.efi.stub: file format not recognized
objcopy: --change-section-vma .initrd=0x0000000003000000 never used
objcopy: --change-section-vma .linux=0x0000000002000000 never used
objcopy: --change-section-vma .cmdline=0x0000000000030000 never used
objcopy: --change-section-vma .osrel=0x0000000000020000 never used
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/usr/lib/python3.11/site-packages/mkosi/__main__.py", line 45, in <module>
    main()
  File "/usr/lib64/python3.11/contextlib.py", line 81, in inner
    return func(*args, **kwds)
           ^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/mkosi/__main__.py", line 41, in main
    run_verb(a)
  File "/usr/lib/python3.11/site-packages/mkosi/__init__.py", line 8254, in run_verb
    manifest = build_stuff(config)
               ^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/mkosi/__init__.py", line 7559, in build_stuff
    image = build_image(state, manifest=manifest)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/mkosi/__init__.py", line 7342, in build_image
    install_unified_kernel(state, root_hash, mount)
  File "/usr/lib/python3.11/site-packages/mkosi/__init__.py", line 4080, in install_unified_kernel
    run(cmd)
  File "/usr/lib/python3.11/site-packages/mkosi/backend.py", line 860, in run
    return subprocess.run(cmdline, check=check, stdout=stdout, stderr=stderr, env={**os.environ, **env}, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib64/python3.11/subprocess.py", line 571, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['objcopy', '--add-section', '.osrel=/var/tmp/mkosi-amxlrk94/root/usr/lib/os-release', '--change-section-vma', '.osrel=0x20000', '--add-section', '.cmdline=/var/tmp/mkosi-amxlrk94/cmdline', '--change-section-vma', '.cmdline=0x30000', '--add-section', '.linux=/var/tmp/mkosi-amxlrk94/root/lib/modules/6.2.0-100.asahi10.1.fc37.aarch64/vmlinuz', '--change-section-vma', '.linux=0x2000000', '--add-section', '.initrd=/var/tmp/mkosi-amxlrk94/root/efi/46d1386bbc7c40baacedc67d2cbab734/6.2.0-100.asahi10.1.fc37.aarch64/initrd', '--change-section-vma', '.initrd=0x3000000', '/var/tmp/mkosi-amxlrk94/root/lib/systemd/boot/efi/linuxaa64.efi.stub', '/var/tmp/mkosi-amxlrk94/root/efi/EFI/Linux/fedora-6.2.0-100.asahi10.1.fc37.aarch64.efi']' returned non-zero exit status 1.`

leifliddy, 2023-03-15:
Are you running this on an `aarch64` system?

leifliddy, 2023-03-15:
If you're running this on an `x86_64` system, try using this branch 
https://github.com/leifliddy/asahi-fedora-usb/tree/grub

kde31, 2023-03-15:
Yes it is an x86_64 system.
Thank you, I will try the link and report if it worked.

kde31, 2023-03-15:
I have run python3 -m pip install --user git+https://github.com/systemd/mkosi.git
I get now unfortunately already at the beginning the error:
`mkosi: error: unrecognized arguments: --repository-directories=mkosi.reposdir`

leifliddy, 2023-03-15:
If you run `which mkosi` does it show the local one?
```
[root@black ~]# which mkosi
/root/.local/bin/mkosi
```

kde31, 2023-03-15:
It looks like this for me:
`[bn@fedora asahi-fedora-usb-grub]$ which mkosi
~/.local/bin/mkosi`

leifliddy, 2023-03-15:
It looks like `mkosi` main changed quite a bit in the last two months. 
I need some time to refactor this script. I'll let you know when I'm done.

kde31, 2023-03-15:
Many thanks already for this.

leifliddy, 2023-03-15:
Ok, I made a few changes to this project. I made the `grub` branch the `main` branch.
That error that you were getting before 
`mkosi: error: unrecognized arguments: --repository-directories=mkosi.reposdir`
was probably due to using `sudo`
If you use `sudo` , the `/usr/bin/mkosi` will be used and not the `~/.local/bin/mkosi` path.

leifliddy, 2023-03-15:
Everything "should" work now.

kde31, 2023-03-16:
But that was fast. Thank you very much for that.
./build.sh required admin rights and the command python3 -m pip install --user git+https://github.com/systemd/mkosi.git installed mkosi into the user profile.
I have in /usr/bin/ mkosi. But I don't know if it is the required version. Maybe I run the above command again without --user?
I'll reload the main branch first and try it out.
But I have time for it only this afternoon

leifliddy, 2023-03-16:
The easiest solution would be to just run both of these commands as root
`python3 -m pip install --user git+https://github.com/systemd/mkosi.git` 
`./build.sh`

kde31, 2023-03-16:
Ok, then I did it wrong.
I omitted --user and installed it. The following warning came up: 
`WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv`
Can I get rid of this with pip uninstall?
Anyway, so with the creation of the USB drive worked.
I haven't booted the Mac Mini from it yet, but I'll test it today and report back then.

kde31, 2023-03-16:
No, I have not yet installed m1n1 and uboot on the Mac Mini.
`Edit the build.sh script and replaced mkosi for the alternate path` is a good idea
As you can see I am a noob.
I have uninstalled mkosi.

kde31, 2023-03-16:
Then I would have a minimal Arch installation on the Mac Mini. Is a good idea.
How much space should I give the partition? 20 GB?

leifliddy, 2023-03-16:
A minimal install should take around 1GB. You could easily go to 5GB if you wanted. But 20GB sounds pretty reasonable + practical as it gives you space to work with. 

Just an FYI, the Asahi Linux project is going to make Fedora an officially supported distro within probably the next month or so.
This topic is being discussed here:
https://app.element.io/#/room/#asahi:fedoraproject.org

In order to future-proof your install, I would install Fedora via 
1.
# note: this hasn't been officially released yet, but it's functional
There's a few minor issuesm but nothing major. 
This is the better install option...as it's supported by Fedora
https://www.asahi-fedora-remix.org/ 

or

2.
# and then there's my project, which is basically only providing mesa + kernel packages at this point.
https://github.com/leifliddy/asahi-fedora-builder
https://leifliddy.com/asahi-linux/37/aarch64/

tbh, I don't see a future for this project once the official Fedora release happens. Well, maybe as an alternate installer...

kde31, 2023-03-16:
Yes, thanks for "taking me by the hand" :-)
Would you now recommend me to wait for the official Fedora release?
I already have a USB drive with Fedora installed. Would it be wise to try this?

leifliddy, 2023-03-19:
I take that back. I just reviewed the Ashai Fedora remix release and it still has a few minor issues to work out. 
I posted about my findings here:
https://app.element.io/#/room/#asahi:fedoraproject.org

So I created a minimal profile profile for that project 
https://pagure.io/fedora-asahi/kiwi-descriptions/c/d569746e810b95e04356eb0b55811cc96eff06f1?branch=rawhide

But that image isn't publicly available yet. 
Which means that when you install Asahi Fedora Remix  you have to choose between either a kde or gnome image. 
So long story short I would just use my project for the time being. 
In anycase I'd be super easy to transition from my project to the Asahi Fedora Remix project in the future. You can do this by simply modifying the repo configs -- removing mine and installing the asahi COPR repos. I'll provide instructions on how to do this once Asahi Fedora Remix is officially released.

kde31, 2023-03-19:
Yes, thank you for the great help and all the information.
But in the end I failed to install m1n1 + uboot.
The Mac Mini M2 is unfortunately not supported yet.
Both attempts ('curl https://alx.sh | sh' and 'curl https://asahi-fedora-remix.org/install | sh' were unsuccessful.

> Am 19.03.2023 um 11:10 schrieb Leif Liddy ***@***.***>:
> 
> But that image isn't publicly available yet.
> Which means that when you install Asahi Fedora Remix you have to choose between either a kde or gnome image.
> So long story short I would just use my project for the time being.
> In anycase I'd be super easy to transition from my project to the Asahi Fedora Remix project in the future. You can do this by simply modifying the repo configs -- removing mine and installing the asahi COPR repos. I'll provide instructions on how to do this once Asahi Fedora Remix is officially released.

leifliddy, 2023-03-19:
Yeah, I bought an mac mini m2 two months ago in the hopes that it would be supported fairly quickly. 
That hasn't happened....probably need to wait another month or so before we see initial support.

kde31, 2023-03-19:
Then we wait and see :-)

closed 2023-11-14 by leifliddy