leifliddy/asahi-fedora-usb #8: No image found on USB drive.

SamilKaraduman, 2023-04-20:
I've managed to install **something** to a USB drive I had laying around, but it simply refuses to boot. Gives me an error message saying that no image was found on the USB drive and to press any key to continue, which obviously didn't do anything. This was all compiled on an x86-64 machine with no hiccups. Perhaps its because the USB drive is formatted as GPT in macOS disk utility?

SamilKaraduman, 2023-04-20:
I'll attach another image shortly.

leifliddy, 2023-04-20:
Ok, I'm building this on an `x86_64` system now. I'll document every step

leifliddy, 2023-04-20:
If you build it a second time you can use the `w` argument 
ie 
```
./build.sh -wd /dev/sdc
```
this saves time as it just wipes the drive vs having to re-partition it.

SamilKaraduman, 2023-04-20:
> If you build it a second time you can use the `w` argument ie
> 
> ```
> ./build.sh -wd /dev/sdc
> ```
> 
> this saves time as it just wipes the drive vs having to re-partition it.

I'll try this. Thanks!

[image]

Heres another picture.

leifliddy, 2023-04-20:
Is it booting?

SamilKaraduman, 2023-04-20:
Nope. Still the same issue. Should I reinstall the whole OS at this point?

leifliddy, 2023-04-20:
Here's what I would try on your system where you're building the usb drive
I ran everything as `root` on an `x86_64` machine. I need to test out procedures running `sudo` 
Since I haven't tested it -- I can't guarantee that using sudo will work at the moment. 
It should work, but you would need to specify the full path to mkosi if it's located in a users directory. 

uninstall `mkosi` just in case you have an old version or something
```
[root@black ~]# pip3 uninstall mkosi
Found existing installation: mkosi 14
Uninstalling mkosi-14:
  Would remove:
    /root/.local/bin/mkosi
    /root/.local/lib/python3.11/site-packages/mkosi-14.dist-info/*
    /root/.local/lib/python3.11/site-packages/mkosi/*
    /root/.local/share/man/man1/mkosi.1
Proceed (Y/n)? y
```
install `mkosi` from main
```
python3 -m pip install --user git+https://github.com/systemd/mkosi.git
```

clone the repo and run the script
``` 
git clone https://github.com/leifliddy/asahi-fedora-usb.git
cd asahi-fedora-usb
./build.sh -d /dev/sdc
```

this is the final part of the log

```
‣   Setting root password
‣   Generating system users
/root/asahi-fedora-usb/.mkosi.tmp5k51je4w/root/usr/lib/sysusers.d/basic.conf:10: conflict with earlier configuration for user 'root', ignoring line.
‣   Applying presets…
Created symlink /root/asahi-fedora-usb/.mkosi.tmp5k51je4w/root/etc/systemd/system/multi-user.target.wants/remote-cryptsetup.target → /usr/lib/systemd/system/remote-cryptsetup.target.
Created symlink /root/asahi-fedora-usb/.mkosi.tmp5k51je4w/root/etc/systemd/system/dbus-org.freedesktop.home1.service → /usr/lib/systemd/system/systemd-homed.service.
Created symlink /root/asahi-fedora-usb/.mkosi.tmp5k51je4w/root/etc/systemd/system/multi-user.target.wants/systemd-homed.service → /usr/lib/systemd/system/systemd-homed.service.
Created symlink /root/asahi-fedora-usb/.mkosi.tmp5k51je4w/root/etc/systemd/system/systemd-homed.service.wants/systemd-homed-activate.service → /usr/lib/systemd/system/systemd-homed-activate.service.
Created symlink /root/asahi-fedora-usb/.mkosi.tmp5k51je4w/root/etc/systemd/system/sysinit.target.wants/systemd-pstore.service → /usr/lib/systemd/system/systemd-pstore.service.
Created symlink /root/asahi-fedora-usb/.mkosi.tmp5k51je4w/root/etc/systemd/system/basic.target.wants/rpmdb-migrate.service → /usr/lib/systemd/system/rpmdb-migrate.service.
Created symlink /root/asahi-fedora-usb/.mkosi.tmp5k51je4w/root/etc/systemd/system/basic.target.wants/rpmdb-rebuild.service → /usr/lib/systemd/system/rpmdb-rebuild.service.
‣   Recording packages in manifest…
‣   Resetting machine ID
‣   Removing random seed
‣   Relabeling files using targeted policy
‣  Unmounting image…
‣ Saving manifest mkosi.rootfs.manifest
‣ Resulting image size is 1.1G.
### Cleaning up
### Mounting usb partitions and copying files
### Setting uuids for partitions in /etc/fstab
### Running systemd-machine-id-setup
Initializing machine ID from random generator.

### Generating EFI bootloader
boot UUID: a1492762-3fe2-4908-a8b9-118439becd26
root UUID: d747cb2a-aff1-4e47-8a33-c4d9b7475df9
Partition: hd0,gpt2
prefix_grub_dir: /grub2

Generating GRUB image...
EFI Bootloader Created
-rwxr-xr-x. 1 root root 577536 Apr 20 22:19 /boot/efi/EFI/BOOT/BOOTAA64.EFI
### Creating BLS (/boot/loader/entries/) entry
/usr/bin/rm: cannot remove '/boot/dtb-6.2.11-450.asahi.fc38.aarch64': Is a directory

### Generating GRUB config
Generating grub configuration file ...
done
### Adding delay to NetworkManager.service
### Enabling system services
### Disabling systemd-firstboot
### Setting selinux to permissive

### Cleanup
### Unmounting usb partitions
### Done
```

It should work at this point. If it doesn't maybe something else is screwed up on your mac.

SamilKaraduman, 2023-04-20:
**sudo ./build.sh -wd /dev/sdb

# # # wiping usb partitions

./build.sh: line 138: mkosi: command not found** 

Thats what I get in return. Why can't just things work? I've reinstalled it, but still nothing.

leifliddy, 2023-04-20:
I said over and over again that I ran the command as the root user and that I can't guarantee that running with sudo will work as the mkosi would need to set statically set. 
And then you run it with sudo and run into that exact issue. 
Can't you just `su` to the `root` user and then those commands like I did?

leifliddy, 2023-04-21:
`sudo` should work, but you need to change this line in `build.sh`
from 
`mkosi` 
to the location was installed with ie `python3 -m pip install --user git+https://github.com/systemd/mkosi.git` 
ie
`/home/leif.liddy/.local/bin/mkosi`

so it'd be something like this 

before
```
mkosi_create_rootfs() {
    mkosi clean
    rm -rf .mkosi*
    mkosi
}
```

after
```
    mkosi_create_rootfs() {
    /home/leif.liddy/.local/bin/mkosi clean
    rm -rf .mkosi*
    /home/leif.liddy/.local/bin/mkosi
}
```
I might create a feature that will simplify that....

SamilKaraduman, 2023-04-21:
The USB now boots. Thank you for your help!

closed 2023-04-21 by SamilKaraduman