leifliddy/asahi-fedora-usb #9: aarch64 is not a valid Architecture

seb4nihel, 2023-06-20:
Hi, I tried to use your script on fedora 38 amd64 but got the following message:
‣ 'aarch64' is not a valid Architecture
Do you have advice for building the boot drive on amd64?
regards Sebastian

seb4nihel, 2023-06-20:
Okay, tested it on a aarch64 host and this has the same result.

leifliddy, 2023-06-20:
I pushed an update recently that should allow you to build the usb drive on both an `x86_64` and `aarch64` system. Building on an `x86_64` system does take a bit of time though. I can't seem to get the usb drive to boot though. 
Something must have changed recently -- as my previously built usb drives don't boot.
I'm conversing with the devs here trying to figure that out 
https://app.element.io/#/room/#asahi:fedoraproject.org

seb4nihel, 2023-06-21:
Thank you! I will give that a try soon, I just need a solution to play around with my broken arch install :)

closed 2023-08-16 by leifliddy