leifliddy/asahi-fedora-usb #13: public key not found error when running ./build.sh

wupgop, 2023-10-16:
I have a MBP 16" with the M2 max chip, installed Fedora 39 with the official installer, and as soon as I was in to Fedora, I followed the instructions in this guide to try to set up WDE:

https://davidalger.com/posts/fedora-asahi-remix-on-apple-silicon-with-luks-encryption/ 

I didn't install any other software, just went right to this. Building the USB drive fails with the following error which is being thrown by mkosi: 

```
Userid     : "@asahi_fedora-remix-branding (None) <@asahi#fedora-remix-branding@copr.fedorahosted.org>"
 Fingerprint: 9E4A FA42 19F4 A0D1 083C B33B A3DC 6751 57D0 48CE
 From       : https://download.copr.fedorainfracloud.org/results/@asahi/fedora-remix-branding/pubkey.gpg
Key imported successfully
fedora                                                                               6.2 kB/s | 6.9 kB     00:01    
Traceback (most recent call last):
  File "/usr/bin/dnf", line 62, in <module>
    main.user_main(sys.argv[1:], exit_code=True)
...
  File "/usr/lib64/python3.12/site-packages/libdnf/repo.py", line 341, in keysFromFd
    return _repo.Key_keysFromFd(fileDescriptor)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
RuntimeError: Public key not found
‣ "/usr/bin/dnf --assumeyes --config=/home/user/asahi-fedora-usb/.mkosi-tmppi95_sxl/pkgmngr/etc/dnf/dnf.conf --best --releasever=38 --installroot=/home/user/asahi-fedora-usb/.mkosi-tmppi95_sxl/root --setopt=keepcache=1 '--setopt=cachedir=/home/user/asahi-fedora-usb/.mkosi-tmppi95_sxl/cache/fedora~38' --setopt=reposdir=/home/user/asahi-fedora-usb/.mkosi-tmppi95_sxl/pkgmngr/etc/yum.repos.d --setopt=varsdir=/home/user/asahi-fedora-usb/.mkosi-tmppi95_sxl/pkgmngr/etc/dnf/vars --setopt=persistdir=/home/user/asahi-fedora-usb/.mkosi-tmppi95_sxl/pkgmngr/var/lib/dnf --setopt=check_config_file_age=0 '--disableplugin=*' --enableplugin=builddep install filesystem" returned non-zero exit code 1.
```

I installed the key at:

```https://download.copr.fedorainfracloud.org/results/@asahi/fedora-remix-branding/pubkey.gpg```

manually, confirmed it was installed, turned off signature checking in /etc/dnf/dnf.conf, installed mkosi both using the method described to install v15.1 from Git, as well as v16 and v18, using the pi install method described in the linked tutorial as well as using dnf, and still receive this error. 

Any help would be great, thanks. LMK any other info needed.

wupgop, 2023-10-17:
Thanks so much for the suggestion and for your work on this project. Oddly enough, when I change releasever to Release=39 in mkosi.conf, it still shows the same error, but with -releasever=39 rather than 38, and the cache directory at fedora~39 rather than 38. 

There are indeed symbolic links at

```
file:///etc/pki/rpm-gpg/RPM-GPG-KEY-fedora-38-aarch64
```
and
```
file:///etc/pki/rpm-gpg/RPM-GPG-KEY-fedora-39-aarch64
```

pointing to

```
file:///etc/pki/rpm-gpg/RPM-GPG-KEY-fedora-38-primary
```
and
```
file:///etc/pki/rpm-gpg/RPM-GPG-KEY-fedora-39-primary
```

Would it be worth re-doing the install, and this time install 38 as the OS instead of 39?

leifliddy, 2023-10-17:
You can also build this usb drive on an x86_64 F38 system as well. It'll just take a bit longer as it'll use qemu to perform the x86_64 --> aarch64 translation. There is a F39 issue out there that may be related to this. 
https://bugzilla.redhat.com/show_bug.cgi?id=2244471

Not sure if that would affect this project as well. 
I'll have time this upcoming weekend to test out building this project on a F39 system.

closed 2023-10-19 by leifliddy

leifliddy, 2023-10-21:
I just updated this project to build a F39 image. Look at the README for more information. You'll need to install mkosi from main to get around that public key issue. Which was resolved by this commit:
https://github.com/systemd/mkosi/commit/eaf9b8bc5f6140b3543f2f09f25dc593f496b95e