leifliddy/asahi-fedora-usb #14: "boot_efi_bootmgr is not defined" error when attempting to boot from the USB drive on M2 mac

wupgop, 2023-10-18:
Howdy. Thanks for your advice in the other issue. 

I went ahead and installed Fedora 38 to get moving forward, and ran into a second issue. I wonder if the necessary syntax has changed in U-Boot 2023.07, which is the version that got installed. See screenshot - when I run the commands to 

```
env set boot_efi_bootmgr
run usb_boot
```

I receive an error "boot_efi_bootmgr is not defined". 

Here's a picture of the screen:

[image]

I tried to see if I could find any documentation of the syntax changing in U-Boot 2023.07 and couldnt find any, however, the documentation says that only M1 macs are support, meaning it could be out of date?

https://u-boot.readthedocs.io/en/latest/board/apple/m1.html

Thanks for any help. 

WG

leifliddy, 2023-10-18:
Interesting. I'm seeing the  same error, but it boots for me on my macbook m1 air running F38 (I just built a new drive).

Here's what I'm seeing

```
** Bad device specification usb -bootable **
Scanning usb :1...
Failed to load '/'
## Error: "boot_efi_bootmgr not defined
Found EFI removable media binary efi/boot/bootaa64.efi
577536 bytes read in 16 ms (34.4 MiB/s)
Booting /efi\boot\bootaa64.efi
```

The first time I booted the usb drive -- it took like 15 seconds or so before it booted. 
It got stuck at that `## Error: "boot_efi_bootmgr not defined` error...but it eventually booted.
Each successive usb boot was super quick -- no hanging
Still seeing that `boot_efi_bootmgr not defined` error msg (very briefly though....I had to take a video and pause it)
Something must have changed recently.   
I'll ask the devs about that error and whether the procedures for booting a usb drive have changed.

leifliddy, 2023-10-18:
@wupgop 
You can also just try adding the usb drive to the grub menu. 
So just boot up your machine into F39, plug in the usb drive, then update grub

```
[root@m1 ~]# grub2-mkconfig -o /boot/grub2/grub.cfg
Generating grub configuration file ...
Found Fedora Linux Asahi Remix 38 (Thirty Eight) on /dev/nvme0n1p6
Found Fedora Linux Asahi Remix 38 (Thirty Eight) on /dev/sda3
done
```

leifliddy, 2023-10-18:
I just edited the README file to include this info.

leifliddy, 2023-10-19:
@wupgop Based on the other issue you submitted -- this issue was likely due to a misconfigured  usb drive. 
Does it boot from u-boot now?

closed 2023-10-20 by leifliddy

wupgop, 2023-10-20:
Yes indeed it does work now.

jvoisin, 2024-03-29:
I can reproduce the issue, getting the `## Error: "boot_efi_bootmgr" not defined` error message.