leifliddy/asahi-fedora-usb #15: build.sh has some issues mounting drives and doesnt write the fedora OS to USB on mbp 16 M2 max

wupgop, 2023-10-19:
Thanks again for all your help.

When trying to add the USB drive to grub, it didn't detect an OS. I looked on the drive and realized it had /boot and /EFI but didnt have a fedora OS as far as I could tell, no squashfs or image either. 

I did the build.sh again and noticed a few errors along the way, I'll paste in my session here and snip out all the parts that went perfectly fine without errors so the errors can be highlighted and not lost in the haystack:

```
[user@computer asahi-fedora-usb]$ git pull
Already up to date.
[user@computer asahi-fedora-usb]$ sudo ./build.sh -wd /dev/sda
[sudo] password for user:
Current working directory: /home/user/asahi-fedora-usb
Path to mkosi: /home/user/.local/bin/mkosi
### The usb drive did not mount
paritioning disk

‣ Removing output files…
‣ Removing output files…
‣ Building default image
Create subvolume '/home/user/asahi-fedora-usb/.mkosi-tmp3jn_0tp7/root'
‣  Copying in package manager file trees…
‣  Mounting image…
‣   Copying in skeleton file trees…
‣   Installing Fedora
Copr repo for fedora-remix-branding owned by @asahi                           6.4 kB/s |  12 kB     00:01
Copr repo for u-boot owned by @asahi                                           12 kB/s |  14 kB     00:01
 ...snip...
Installing       : libbpf-2:1.1.0-2.fc38.aarch64                                                    152/400
  Running scriptlet: xml-common-0.6.3-60.fc38.noarch                                                  153/400 
 ...snip...
  Running scriptlet: kernel-16k-core-6.5.6-403.asahi.fc38.aarch64                                     400/400
grub2-probe: error: cannot find a device for / (is /dev mounted?).
No path or device is specified.
Usage: grub2-probe [OPTION...] [OPTION]... [PATH|DEVICE]
Try 'grub2-probe --help' or 'grub2-probe --usage' for more information.
dracut: WARNING: running in hostonly mode in a container!!
dracut: No '/dev/log' or 'logger' included for syslog logging

  Running scriptlet: kernel-16k-modules-6.5.6-403.asahi.fc38.aarch64                                  400/400 
  Running scriptlet: rootfiles-8.1-33.fc38.noarch                                                     400/400 
  Running scriptlet: asahi-repos-1-10.noarch                                                          400/400 
Creating group 'systemd-coredump' with GID 998.
Creating user 'systemd-coredump' (systemd Core Dumper) with UID 998 and GID 998.
Creating group 'systemd-resolve' with GID 193.
Creating user 'systemd-resolve' (systemd Resolver) with UID 193 and GID 193.
Creating group 'systemd-timesync' with GID 997.
Creating user 'systemd-timesync' (systemd Time Synchronization) with UID 997 and GID 997.

mount: /run/.system-efi: can't find PARTUUID=5bfd3338-f75e-4f7a-8d20-96242d5276e2.

  Verifying        : asahi-repos-1-10.noarch                                                            1/400 
  Verifying        : kernel-16k-6.5.6-403.asahi.fc38.aarch64                                            2/400 
 ...snip...
  Verifying        : xxhash-libs-0.8.2-1.fc38.aarch64                                                 399/400
  Verifying        : zchunk-libs-1.3.2-1.fc38.aarch64                                                 400/400 

Installed:
  NetworkManager-1:1.42.8-1.fc38.aarch64                NetworkManager-libnm-1:1.42.8-1.fc38.aarch64          
  NetworkManager-wifi-1:1.42.8-1.fc38.aarch64           acl-2.3.1-6.fc38.aarch64                              
 ...snip...
  zchunk-libs-1.3.2-1.fc38.aarch64                      zlib-1.2.13-3.fc38.aarch64                            
  zram-generator-1.1.2-4.fc38.aarch64                   zram-generator-defaults-1.1.2-4.fc38.noarch           

Complete!
‣   Generating system users
‣   Applying presets…
Removed "/home/user/asahi-fedora-usb/.mkosi-tmp3jn_0tp7/root/etc/systemd/system/systemd-homed.service.wants/systemd-homed-activate.service".
Created symlink /home/user/asahi-fedora-usb/.mkosi-tmp3jn_0tp7/root/etc/systemd/system/systemd-homed.service.wants/systemd-homed-activate.service → /usr/lib/systemd/system/systemd-homed-activate.service.
‣   Applying first boot settings
/home/user/asahi-fedora-usb/.mkosi-tmp3jn_0tp7/root/etc/passwd written
/home/user/asahi-fedora-usb/.mkosi-tmp3jn_0tp7/root/etc/shadow written.
‣   Generating hardware database
‣   Recording packages in manifest…
‣   Relabeling files using targeted policy
‣  Unmounting image…
‣  Saving manifest mkosi.rootfs.manifest
‣  /home/user/asahi-fedora-usb/mkosi.rootfs size is 1.1G.
### Cleaning up
### Mounting usb partitions and copying files
mount: mnt_usb: can't find UUID="d747cb2a-aff1-4e47-8a33-c4d9b7475df9".
[user@computer asahi-fedora-usb]$ cd /run/media/user/KRD
[user@computer KRD]$ du -h
2.8M    ./EFI/boot
2.8M    ./EFI
1.3M    ./System/Library/CoreServices
1.3M    ./System/Library
1.3M    ./System
176K    ./boot/grub/cfg
1.4M    ./boot/grub/fonts
48K     ./boot/grub/i386-efi/cfg
5.6M    ./boot/grub/i386-efi
48K     ./boot/grub/i386-pc/cfg
5.4M    ./boot/grub/i386-pc
208K    ./boot/grub/locale
176K    ./boot/grub/pxe
16K     ./boot/grub/roms
1.3M    ./boot/grub/themes/kl
3.2M    ./boot/grub/themes/starfield
4.5M    ./boot/grub/themes
80K     ./boot/grub/x86_64-efi/cfg
11M     ./boot/grub/x86_64-efi
65M     ./boot/grub
65M     ./boot
536M    ./data
48K     ./System Volume Information
608M    .
[user@computer KRD]$ ls */*
 data/000-core.srm          data/003-kl.srm             data/008-firefox.srm
 data/000-core.srm.sha512   data/003-kl.srm.sha512      data/008-firefox.srm.sha512
 data/001-xorg.srm          data/004-krt.srm            data/kernel.dat
 data/001-xorg.srm.sha512   data/004-krt.srm.sha512     data/kernel.dat.sha512
 data/002-xfce.srm          data/005-bases.srm         'System Volume Information/IndexerVolumeGuid'
 data/002-xfce.srm.sha512   data/005-bases.srm.sha512  'System Volume Information/WPSettings.dat'

boot/grub:
cfg    grub.cfg      grubenv      i386-efi  initrd.xz      k-x86     k-x86_64.sig  locale  roms    x86_64-efi
fonts  grub.cfg.sig  grubenv.sig  i386-pc   initrd.xz.sig  k-x86_64  k-x86.sig     pxe     themes

EFI/boot:
boot.efi  bootia32.efi  bootx64.efi  krd_grub.efi

System/Library:
CoreServices
[user@computer KRD]$
```

So the first obvious issue is:

```$ sudo ./build.sh -wd /dev/sda
[sudo] password for user:
Current working directory: /home/user/asahi-fedora-usb
Path to mkosi: /home/user/.local/bin/mkosi
### The usb drive did not mount
```

Then later on a few messages about not being able to mount devices, followed by the final failure to copy the OS files:

```
### Cleaning up
### Mounting usb partitions and copying files
mount: mnt_usb: can't find UUID="d747cb2a-aff1-4e47-8a33-c4d9b7475df9".
```

Sorry to keep posting issues, I'm just really looking forward to getting this working :)

Thank you for your dedication to this project and your help.

leifliddy, 2023-10-19:
It's difficult to make the script sudo compatible as I would need to prefix each privileged command with sudo. You need to run the script as root. I'll update the readme later with that. What does the output of ```blkid``` show for ```/dev/sda```? It sound like the drive was never partitioned by this script. Which means you shouldn't be using the -w option. I would plugin the usb drive, then ensure that the drive isn't mounted anywhere (like under /run/media), and then run ```./build.sh -d /dev/sda``` as root

wupgop, 2023-10-19:
Thanks, running as root, and without the -w, did the trick. Sorry for missing the -w parameter issue, I started out originally with a different USB drive that I did format with the build script, but in this example I moved to a different USB drive in case the drive was faulty and forgot to remove the w. But it's up and running now, fantastic! Thanks so much for this tool and your help.

closed 2023-10-20 by leifliddy