leifliddy/asahi-fedora-usb #16: Boot issue with fed39

julesGoullee, 2023-11-09:
I have installed on my internal drive fed39, trying to boot on usb, I can't boot on it. During the build setup of the usb device, I've spotted an error (see screenshot 1 bellow), when trying with uboot cmd is doesnt detect the usb device, I added the entry to grub (successfully) from my internal fed with the usb mount, when I try to boot from the grub I got this error (screenshot 2), then if I try to boot on my internal feb from the first grub entry it saw up almost the same error I need to reboot completly then I can boot on my internal
[image]
[image]

leifliddy, 2023-11-09:
I try and test this every week or so. I really don't like installing from mkosi from main as it changes so fast. I should probably peg it to a specific commit until they cut a new release. With that said I'm not sure what your're issue it I've never seen that /run/.system-efi issue before.It looks like it's trying to mount your internal efi partition -but it can't for some reason. I'll have some time tomorrow to help you troubleshoot this further.

leifliddy, 2023-11-14:
I just made a number or improvements. Can you try and build the drive again and let me know if it works?

julesGoullee, 2023-11-14:
All good now 😊

1Jo1, 2023-12-15:
@leifliddy I have similar issue I think(last commit fa11840)
[image]

leifliddy, 2023-12-15:
@1Jo1 I just built the drive and it boots fine. 
The last commit was a README file edit, it wouldn't have affect the drive.

I need some more details to troubleshoot your issue:

Can you post the following output? (substitute `/dev/sda` with your usb device name)
```
[root@m1 ~]# blkid -s UUID -o value /dev/sda[1-3]
3051-D434
a1492762-3fe2-4908-a8b9-118439becd26
d747cb2a-aff1-4e47-8a33-c4d9b7475df9
````
If you don't see the exact output, then rebuild the drive without the `-w` option.
ie
`./build -d /dev/sda`

Are you building this on a `F38` or `F39` system?
Are you building this on an `aarch64` system or a `x86_64` system

What version of mkosi are you using (should be `v19`)? 
Can you post the output of these commands (on the system that you're building the drive with)?
```
[root@m1 ~]# which mkosi
/usr/bin/mkosi

[root@m1 ~]# mkosi --version
mkosi 19
```

How are you booting the drive? via u-boot cmds or via (internal) grub?

1Jo1, 2023-12-15:
```
fedora@jo ~]# blkid -s UUID -o value /dev/sda[1-3]
```
I dont see any output...even when try` ./build -d /dev/sda`
I'm using F39 aarch64 (M2 Pro), I'm gonna try to build it on `x86_64` when I'm at home

```
[fedora@jo ~]# which mkosi
/usr/bin/mkosi

[fedora@jo ~]# mkosi --version
mkosi 19
```
first I tried via u-boot but didnt detect my usb drive, so I tried that via (internal) grub

leifliddy, 2023-12-15:
Do you see your usb device in the `lsblk` output?
```
[root@m1 ~]# lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
sda           8:0    1  59.8G  0 disk 
├─sda1        8:1    1   600M  0 part 
├─sda2        8:2    1     1G  0 part 
└─sda3        8:3    1  58.2G  0 part 
```

1Jo1, 2023-12-16:
https://gist.github.com/1Jo1/2e4f70fc20a3d982ae1cfb3a304a6ed6 here the logs
```
fedora@joasahi-fedora-usb   main lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
sda           8:0    1  14.4G  0 disk 
├─sda1        8:1    1   600M  0 part 
├─sda2        8:2    1     1G  0 part 
└─sda3        8:3    1  12.8G  0 part 
```
same output

leifliddy, 2023-12-16:
are you running `./build -d /dev/sda` as `root`?
I should probably enforce that in the script....

Can you post the full `blkid | grep /dev/sda` output?

1Jo1, 2023-12-16:
> are you running `./build -d /dev/sda` as `root`? I should probably enforce that in the script....
no as `sudo` I will try that as root now
> Can you post the full `blkid | grep /dev/sda` output?
sorry my bad just noticed I forgot the `sudo` `fedora@jo ~]# blkid -s UUID -o value /dev/sda[1-3]`
```
fedora@jo~ sudo blkid | grep /dev/sda  
/dev/sda2: LABEL="fedora-usb-boot" UUID="a1492762-3fe2-4908-a8b9-118439becd26" BLOCK_SIZE="4096" TYPE="ext4" PARTLABEL="Linux filesystem" PARTUUID="1aba7cb0-1e28-4ea4-9816-d447dd3e07a0"
/dev/sda3: LABEL="fedora-usb-root" UUID="d747cb2a-aff1-4e47-8a33-c4d9b7475df9" BLOCK_SIZE="4096" TYPE="ext4" PARTLABEL="Linux filesystem" PARTUUID="43560be4-00b5-4d35-8a5f-371ac75dbd49"
/dev/sda1: LABEL_FATBOOT="EFI-USB-FED" LABEL="EFI-USB-FED" UUID="3051-D434" BLOCK_SIZE="512" TYPE="vfat" PARTLABEL="EFI system partition" PARTUUID="065b647a-4e8d-40e8-89e5-51a68561b77d"
```

leifliddy, 2023-12-16:
Running scripts that contain tons of privileged commands is a PITA to get to run with sudo. Because unlike a binary file, each individual privileged command within a script needs to be prefix with sudo. 

Ok, I'll add the logic to the script now to ensure users are running it as root.....

1Jo1, 2023-12-16:
> Running scripts that contain tons of privileged commands is a PITA to get to run with sudo. Because unlike a binary file, each individual privileged command within a script needs to be prefix with sudo.
> 
> Ok, I'll add the logic to the script now to ensure users are running it as root.....

true that make sense, I just built the script as root, still the same issue...I will try different usb stick just be safe haha

leifliddy, 2023-12-16:
Try running these command in the uboot shell/prompt

```
usb reset
usb info
usb storage
```
Does it see your usb device?

1Jo1, 2023-12-16:
[image]
so weird...no storage device

leifliddy, 2023-12-16:
What model/system is this? A mac mini?

leifliddy, 2023-12-16:
Odd, I'm seeing the same issue with my usb drive. I normally just boot from it via the main grub menu (see readme file for more info on that). It's odd that it's not seeing the device from uboot. Maybe something changed with that last uboot update or maybe it's just flaky.

1Jo1, 2023-12-16:
> What model/system is this? A mac mini?

macbook pro 14 inch 2023

leifliddy, 2023-12-16:
What!? I can't boot boot it from grub either....one sec. I'm going to reinstall this drive....

1Jo1, 2023-12-16:
> Odd, I'm seeing the same issue with my usb drive. I normally just boot from it via the main grub menu (see readme file for more info on that). It's odd that it's not seeing the device from uboot. Maybe something changed with that last uboot update or maybe it's just flaky.

I'm using a usb c/usb a adapter for my usb stick, is that the reason maybe?

leifliddy, 2023-12-16:
No, I'm using the same adapter.

1Jo1, 2023-12-16:
> What!? I can't boot it from grub either....one sec. I'm going to reinstall this drive....

interesting, you didnt change much last month, so its probably one of the packages on f39 which causes this issue I guess?

leifliddy, 2023-12-16:
Something odd is happening, it was working fine this morning. I've upgrade a few packages since then. Not sure if that's the cause or not. I need some time to test a few things out and to communicate with the asahi devs as well.....

leifliddy, 2023-12-16:
See if you can join this channel
https://app.element.io/#/room/#asahi:fedoraproject.org

I sorted out the issue....well sort of. The issue is  either with the latest kernel or with m1n1. 
If you cd to `/boot/efi/m1n1` and rename `boot.bin.old` to `boot.bin` it should work. 
backup the original `boot.bin` just in case. Discussing this issue with the devs now....

1Jo1, 2023-12-16:
> See if you can join this channel https://app.element.io/#/room/#asahi:fedoraproject.org
>
> I sorted out the issue....well sort of. The issue is either with the latest kernel or with m1n1. If you cd to [...]

I tried to rename `boot.bin.old`, but it didnt work, the file size `boot.bin.old` is the same as `boot.bin` as I installed fresh fedora asahi linux few hours ago

leifliddy, 2023-12-16:
Can you try this. 
Remove the current version of m1n1 with `rpm -e --nodeps m1n1`
Then install the older version of m1n1
`rpm -ivh https://leifliddy.com/asahi-linux/39/aarch64/m1n1-1.4.8-1.fc39.aarch64.rpm`
then run `update-m1n1` and see if that works.

1Jo1, 2023-12-16:
> Can you try this? Remove the current version of m1n1 with `rpm -e --nodeps m1n1` Then install the older version of m1n1 `rpm -ivh https://leifliddy.com/asahi-linux/39/aarch64/m1n1-1.4.8-1.fc39.aarch64.rpm` then run `update-m1n1` and see if that works.

it didn't work, I reinstall it via `build.sh -d /dev/sda`, my usb stick is still missing in u-boot and via grub is the same error

leifliddy, 2023-12-16:
Thanks for trying. I'm sure this will get sorted out. I'll ask marcan to look into this....

leifliddy, 2023-12-17:
This user was having the same issue
https://github.com/leifliddy/asahi-fedora-builder/issues/30

This commit should fix the issue
https://github.com/AsahiLinux/m1n1/commit/90eef7223e5da9bdc7ad7f823e7748326ba862d2

`m1n1-1.4.11` should be released soon.

1Jo1, 2023-12-18:
Thanks!
I'm gonna try it later
BTW I can't build anymore as root, I think its this commit https://github.com/leifliddy/asahi-fedora-usb/commit/51a32177777b8cd6f0b856317debc8bc8ed00362

leifliddy, 2023-12-18:
That was a typo, I should have enclosed those conditional statements with parenthesis. 
I just pushed out a new change. Can you try again?

leifliddy, 2023-12-18:
It should be good now, can you please try and send me your feedback?

1Jo1, 2023-12-29:
I'm sorry I was in vacation the last two weeks, couldn't test it...had only my thinkpad...
I tried multiple times, it seems that on my M2 Pro the usb device in uboot doesn't show up at all, never did...I tried with a M1 that works perfectly with the same usb device
both have the same m1n1, kernel and uboot version(newest version all)