leifliddy/asahi-fedora-usb #19: Running build.sh leaves Fedora unbootable

rlopzc, 2023-12-31:
Hello.

I've run the commands to create a bootable USB, which leaves Fedora unbootable with the following message when choosing Fedora from Grub.

```
error
/grub-core/comnands/search.c:315:no such device:
a1492762-3fe2-4908-a8b9-118439becd26.
error: //grub-core/kern/fs.c:121:unknown filesystem.
error: /1/grub-core/loader/arm64/1inux.c:282:you need to load the kernel
first,
Press any key to continue...
```

leifliddy, 2023-12-31:
That sounds a lot like this issue
https://github.com/leifliddy/asahi-fedora-usb/issues/16

Are you running `m1n1-1.4.11` on your internal drive?

If you run `usb storage` in u-boot -- does it show you the information on your usb device?

rlopzc, 2024-01-01:
`usb storage` doesn't show the usb information.

I'm running the latest. I just installed fedora and then tried to create the bootable CD. How can I verify that I'm running `m1n1-1.4.11`?

leifliddy, 2024-01-01:
Just run `rpm -q m1n1`
ie
```
[root@m1 ~]# rpm -q m1n1
m1n1-1.4.11-1.fc39.aarch64
```

The `m1n1-1.4.10` version had an issue where usb devices were not seen by uboot -- and `m1n1-1.4.11` resolved that. 

Just perform a `dnf upgrade` to upgrade your packages (if you haven't already)

BTW, what hardware are you running?

rlopzc, 2024-01-01:
I need to reinstall Fedora to run those commands (`rpm`, `dnf`). I'll try it this week.

The hardware is a MBP 14-inch M1 Pro.

The USB drive is type A, using a U-Green type-c adapter.

I've run those commands previously, and they didn't make any difference.

leifliddy, 2024-01-01:
Why would you need to reinstall Fedora?
Can't you just boot from the internal drive and run that rpm command to find which version of m1n1 you have installed? Or can you not boot from the internal drive? If not, I can show you how to rescue it without needing to reinstall.

rlopzc, 2024-01-01:
Yeah, I can’t boot the internal fedora. Both grub entries doesn’t work with
similar error messages from both.

Can you show me how? Thanks

leifliddy, 2024-01-02:
Sure, basically you're just going to use this project.
https://github.com/leifliddy/fedora-macos-vagrant-builder

There's already a pre-built that I've created. So what you need to do is boot into macOS

run the following two commands to install qemu/vagrant/vagrant -qemu plugin 
````
brew install qemu vagrant
vagrant plugin install vagrant-qemu
````
Also post the output of this command to see which version of qemu you have installed
```
[leif.liddy@m1 ~]$ qemu-system-aarch64 --version
QEMU emulator version 8.2.0
```
According to user `noisycoil`, version 8.2.0 is broken and he needed to downgrade the version . I'm installing this on my system now to see if that's true. 

next just run these commands 
```
mkdir vagrant-fedora
cd vagrant-fedora
curl https://leifliddy.com/vagrant.sh | sh
```
It'll prompt you for your sudo password a few times, but afterwards it should install and run the Fedora vagrant box. 

this cmd will show you if it installed the vagrant box and well as the version
```
[leif.liddy@m1 ~]$ vagrant box list
fedora_39      (libvirt, 20231208)
```

Then just sure you're in that `vagrant-fedora` directory and then run  `vagrant ssh fedora` to ssh into the vagrant box. 
If that works, then just run `chroot.asahi` to mount your internal Linux partitions and chroot into you internal environment. 

Let me know if you get this far and I'll show how to rescue the system from there. I'm running these cmds now to see if there is an issue with qemu v8.2.0
I just need to few minutes to see if there's an issue or not.....

leifliddy, 2024-01-02:
oh god, there something definitely wrong. 

```
[leif.liddy@m1 vagrant-fedora]$ sudo ./serial-connect.sh
BdsDxe: loading Boot0001 "UEFI Misc Device" from PciRoot(0x0)/Pci(0x3,0x0)
BdsDxe: starting Boot0001 "UEFI Misc Device" from PciRoot(0x0)/Pci(0x3,0x0)

Synchronous Exception at 0x00000000BC562000
PC 0x0000BC562000
PC 0x0000BC5C4288
PC 0x0000BC5C4338
PC 0x0000BC5C51B4
PC 0x0000BC5C2030
PC 0x000047686788 (0x00004767F000+0x00007788) [ 1] DxeCore.dll
PC 0x0000BFCCAECC (0x0000BFCC4000+0x00006ECC) [ 2] BdsDxe.dll
PC 0x0000BFCCDFD4 (0x0000BFCC4000+0x00009FD4) [ 2] BdsDxe.dll
PC 0x00004768A00C (0x00004767F000+0x0000B00C) [ 3] DxeCore.dll
[ 1] /home/kraxel/projects/qemu/roms/Build/ArmVirtQemu-AARCH64/DEBUG_GCC5/AARCH64/MdeModulePkg/Core/Dxe/DxeMain/DEBUG/DxeCore.dll
[ 2] /home/kraxel/projects/qemu/roms/Build/ArmVirtQemu-AARCH64/DEBUG_GCC5/AARCH64/MdeModulePkg/Universal/BdsDxe/BdsDxe/DEBUG/BdsDxe.dll
[ 3] /home/kraxel/projects/qemu/roms/Build/ArmVirtQemu-AARCH64/DEBUG_GCC5/AARCH64/MdeModulePkg/Core/Dxe/DxeMain/DEBUG/DxeCore.dll
....
````

I need a few minutes to sort this out......

leifliddy, 2024-01-02:
It's related to the firmware version it's using 
This project is having the same issue with qemu 8.2.0
https://github.com/lima-vm/lima/pull/2061
looking into it now.....

leifliddy, 2024-01-02:
ha, I got it to work with the new firmware. I just need sometime to test it out a bit more and I'll post instructions on how to upgrade the edk2 firmware

leifliddy, 2024-01-02:
I reworked the entire firmware logic -- everything should work find with the latest brew version of qemu

you should just need to do the following to chroot into your internal Linux drive
```
mkdir vagrant-fedora
cd vagrant-fedora
curl https://leifliddy.com/vagrant.sh | sh
./vagrant ssh
chroot.asahi
```
Let me know if you experience any issues with that.

rlopzc, 2024-01-02:
Thanks @leifliddy I'll try this as soon as I can!

closed 2024-01-11 by leifliddy