leifliddy/asahi-fedora-usb #20: "You must run this script as root" even when running as root

LilithHafner, 2024-01-28:
I got this error:
```
x@fedora:~$ git clone https://github.com/leifliddy/asahi-fedora-usb
Cloning into 'asahi-fedora-usb'...
remote: Enumerating objects: 861, done.
remote: Counting objects: 100% (316/316), done.
remote: Compressing objects: 100% (200/200), done.
remote: Total 861 (delta 189), reused 200 (delta 106), pack-reused 545
Receiving objects: 100% (861/861), 9.62 MiB | 8.86 MiB/s, done.
Resolving deltas: 100% (438/438), done.
x@fedora:~$ cd asahi-fedora-usb/
x@fedora:~/asahi-fedora-usb$ su
Password: 
root@fedora:/home/x/asahi-fedora-usb# ./build.sh -d /dev/sda
You must run this script as root
root@fedora:/home/x/asahi-fedora-usb# whoami
root
root@fedora:/home/x/asahi-fedora-usb# ./build.sh -d /dev/sda
You must run this script as root
root@fedora:/home/x/asahi-fedora-usb# sudo ./build.sh -d /dev/sda
### Preparing USB device
GPT fdisk (gdisk) version 1.0.9

Partition table scan:
  MBR: protective
  BSD: not present
  APM: not present
  GPT: present

Found valid GPT with protective MBR; using GPT.
...
```

It seems that I needed to use sudo as root? That seems excessive and redundant.

leifliddy, 2024-01-29:
If you `su` with the dash `-` /(ie `su -`)
Then the `USER` variable is not going to resolve to `root`
```
[leif.liddy@mac ~]$ su 

[root@mac leif.liddy]# echo $USER
leif.liddy
```

Ok I just changed the logic to use `whoami` to identify `root` instead of the USER variable.

LilithHafner, 2024-01-29:
Perfect, thanks!

closed 2024-01-29 by LilithHafner