leifliddy/asahi-fedora-usb #21: "Do you want to proceed?" prompt is broken

LilithHafner, 2024-01-28:
When running the build script `sudo ./build.sh -d /dev/sda`, I saw this in the output

```
...
Command (? for help): Partition number (3-128, default 3): First sector (34-488914910, default = 3328000) or {+-}size{KMGTP}: Last sector (3328000-488914910, default = 488912895) or {+-}size{KMGTP}: Current type is 8300 (Linux filesystem)
Hex code or GUID (L to show codes, Enter = 8300): Changed type of partition to 'Linux filesystem'

Command (? for help): 
Final checks complete. About to write GPT data. THIS WILL OVERWRITE EXISTING
PARTITIONS!!

Do you want to proceed? (Y/N): OK; writing new GUID partition table (GPT) to /dev/sda.
The operation has completed successfully.
mkfs.fat 4.2 (2021-01-31)
mke2fs 1.47.0 (5-Feb-2023)
/dev/sda2 contains a ext4 file system labelled 'fedora-usb-boot'
	created on Sun Jan 28 17:35:05 2024
Creating filesystem with 262144 4k blocks and 65536 inodes
Filesystem UUID: a1492762-3fe2-4908-a8b9-118439becd26
Superblock backups stored on blocks: 
	32768, 98304, 163840, 229376
...
```

But I never had an opportunity to respond to the "Do you want to proceed?" prompt. It printed simultaneously with the subsequent "OK; ...".

leifliddy, 2024-01-29:
If you look at the logic -- it's piping the gdisk subcommands to gdisk. You can't get an interactive prompt that way.
```
echo -e 'o\ny\nn\n\n\n+600M\nef00\nn\n\n\n+1G\n8300\nn\n\n\n\n8300\nw\ny\n' | gdisk $usb_device
```

You can easily modify the script to suit your needs, or just manually partition your usb drive -- and comment your that function.

LilithHafner, 2024-01-29:
That makes sense as an explanation for why the situation is the way it is. It would still be nice if either the user was actually warned before erasing the drive (possibly through a separate prompt) or if the prompt was silenced.

I imagine that this is a low priority issue.