leifliddy/asahi-fedora-usb #22: Installer failed

LilithHafner, 2024-01-29:
When running the installer on an M2 mac with asahi, I ran into the following (apparently critical) error:

```
root@fedora:/home/x/asahi-fedora-usb# sudo ./build.sh -d /dev/sda
### Preparing USB device
GPT fdisk (gdisk) version 1.0.9
...
Error: Transaction failed
‣ "env HOME=/ dnf --assumeyes --best --releasever=39 --installroot=/var/tmp/mkosi-workspacervire43w/root --setopt=keepcache=1 --setopt=cachedir=/home/x/asahi-fedora-usb/mkosi.cache/cache/dnf --setopt=persistdir=/home/x/asahi-fedora-usb/mkosi.cache/lib/dnf --setopt=install_weak_deps=0 --setopt=check_config_file_age=0 '--disableplugin=*' --enableplugin=builddep --config=/etc/dnf/dnf.conf --setopt=reposdir=/etc/yum.repos.d --setopt=varsdir=/etc/dnf/vars install NetworkManager NetworkManager-wifi acl alsa-ucm-asahi arch-install-scripts asahi-audio asahi-bless asahi-fwupdate asahi-nvram asahi-platform-metapackage asahi-repos asahi-scripts audit bash-completion binutils btrfs-progs cloud-utils-growpart cracklib cracklib-dicts cryptsetup dhcp-client diffutils dnf dnf-plugins-core dosfstools dracut-asahi e2fsprogs efivar findutils grub2-efi-aa64 grub2-efi-aa64-modules grubby hostname iproute iputils kernel-16k kernel-16k-modules-extra kmod light linux-firmware man-db man-pages ncurses net-tools openssh-clients openssh-server passwd pciutils rootfiles rpm rsync selinux-policy shim-aa64 speakersafetyd sudo systemd-resolved tar tiny-dfr usbutils vim-enhanced vim-minimal wget wpa_supplicant zram-generator zram-generator-defaults" returned non-zero exit code 1.
```

Full error: https://gist.github.com/LilithHafner/872bf05baa66d1b1ec39c04fc8f6e808

Happy to provide more system info or run troubleshooting as needed.

leifliddy, 2024-01-29:
You're getting this error over and over again
`rename failed - No space left on device`
It looks like you ran out of space. How full is your hdd?

LilithHafner, 2024-01-29:
Thanks! That is almost certainly the problem. I only have 2.1GB free. Ironically, the reason I'm using this at all is to try to resize my asahi root partition so that I have more free space. I'll try again once I get my hands on more free disk space.