leifliddy/asahi-fedora-usb #23: Script fails if SELinux is disabled

div72, 2024-01-31:
Script fails with `setenforce: SELinux is disabled` on the following line if the SELinux is disabled:
https://github.com/leifliddy/asahi-fedora-usb/blob/9b72a67e9f5a545d069dbdf67cf767b003943274/build.sh#L250

Might want to check if it's in the Enforcing state before setting it to Permissive. (`getenforce` works and returns `Disabled`)

leifliddy, 2024-01-31:
Good point. I should check to see if selinux disabled or not. I'll make that change tonight...

leifliddy, 2024-02-01:
I just pushed the following commit
https://github.com/leifliddy/asahi-fedora-usb/commit/1f04f2880954f9cc33e5b0329b899d64b0fc7c82

Can you tell me if that works for you?

div72, 2024-02-01:
Yep, fixed. Thanks.

closed 2024-02-01 by div72