leifliddy/asahi-fedora-usb #26: Error: Loading repository

Gebbx, 2024-03-27:
Hi, have this trouble:

[image]

Thanks for helping!

leifliddy, 2024-03-27:
hmmm...not sure why it's not working for you. 

```
[root@m1 asahi-fedora-usb]# ./build.sh -wd /dev/sda
### Wiping usb partitions
‣ Removing output files…
‣ Removing output files…
‣ Building default image
Create subvolume '/root/.cache/mkosi-workspacerv_oxp13/root'
‣  Copying in package manager file trees…
‣  Copying in skeleton file trees…
‣  Installing Fedora
Updating and loading repositories:
 Copr repo for fedora-remix-branding owned by @asahi                          100% |   3.2 KiB/s |   1.5 KiB |  00m00s
 Copr repo for fedora-remix-scripts owned by @asahi                           100% |  10.0 KiB/s |   1.5 KiB |  00m00s
 Copr repo for kernel owned by @asahi                                         100% |  10.0 KiB/s |   1.5 KiB |  00m00s
 Copr repo for u-boot owned by @asahi                                         100% |   9.8 KiB/s |   1.5 KiB |  00m00s
 fedora                                                                       100% |  20.0 KiB/s |  15.9 KiB |  00m01s
 updates                                                                      100% |  24.0 KiB/s |  11.0 KiB |  00m00s
 Copr repo for kernel owned by @asahi                                         100% |  23.7 MiB/s |  54.7 MiB |  00m02s
 updates                                                                      100% |  11.7 MiB/s |  20.0 MiB |  00m02s
Repositories loaded.
Package                                    Arch    Version                            Repository                  Size
Installing:                                                                                                           
 filesystem                                aarch64 3.18-6.fc39                        fedora                 106.0   B
```

The repos this project uses are located here. 
https://github.com/leifliddy/asahi-fedora-usb/tree/main/mkosi.skeleton/etc/yum.repos.d

Can you not access this url?
https://download.copr.fedorainfracloud.org/results/@asahi/kernel/fedora-39-aarch64/

Gebbx, 2024-03-30:
Succeeded on the third try. Perhaps the problem arose due to the "live" OS. After installation there were no such problems.

closed 2024-03-30 by Gebbx