leifliddy/asahi-fedora-usb #28: can't find UUID error

kevinvarley, 2024-04-04:
Running `./build.sh -wd /dev/sda` results in the following error:

`### Cleaning up
### Mounting usb partitions and copying files
mount: /home/kevin/Downloads/asahi-fedora-usb/mnt_usb: can't find UUID="d747cb2a-aff1-4e47-8a33-c4d9b7475df9".`

mkosi v20.1
Fedora 39

Any ideas?

kevinvarley, 2024-04-04:
Wiping the drive with the KDE disk formatter GUI (new partition table and partition) then, recloned the repo and tried again and it all worked.

closed 2024-04-04 by kevinvarley

leifliddy, 2024-04-04:
You need to remove the `-w` argument
```./build.sh -d /dev/sda```

The `-w` arg is only meant to be used if you've already flashed the image to this drive -- and hence it'll have these UUID's (and associated filesystems already created. 
```
EFI_UUID='3051-D434'
BOOT_UUID='a1492762-3fe2-4908-a8b9-118439becd26'
ROOT_UUID='d747cb2a-aff1-4e47-8a33-c4d9b7475df9'
```

For new drives -- don't use the `-w` option