leifliddy/asahi-fedora-usb #33: Usb fedora stuck during boot

ObelixLeGaulois, 2024-07-24:
Hi, this was working last week, but it seems that now it does not. When booting the usb key, the process stops and the system get stuck on the last line of the screenshot attached.
Then after 20secondes the system make an hard-reset.
Not sure what is happening ....
[image]

a-plastic-bag, 2024-08-02:
Reproduced here on my 2021 M1 Pro MacBook Pro. Live environments seem to be my only way to restore my btrfs backups, so without this I can’t fix my Asahi install :(

ImanolBarba, 2024-08-08:
Same issue :(

ImanolBarba, 2024-08-08:
I was able to boot by telling grub to load the kernel and initramfs from the working asahi install on the disk (not the one on USB), so either the kernel cmdline or something related to kernel/initramfs installed in the USB is wrong.

danacr, 2024-08-15:
Same here @ImanolBarba can you share the commands that got it working?

ImanolBarba, 2024-08-16:
1. Boot with the USB drive
2. Once in GRUB2, input
```
set root=(hdX, Y) 
linux /vmlinuz-ABC ro root=/dev/sdZ verbose
initrd /initramfs-ABC
boot
```

(I'm AFK and I wrote this from memory, I think it should work but I might be forgetting something) 

Where `X` is the disk where asahi /boot is, Y is the partition. `ls` command will list disks

`Z` Is the expected drive name for the usb, if you have no other storage media should be `sda`

`ABC` is the kernel version and release. GRUB2 supports listing files and tab completion. 

This is a shitty workaround, not a proper solution, but can get you out in a pinch (for me it was the need to resize the asahi partition on disk). Essentially what we're doing here is borrowing the main install kernel and initramfs to boot the Linux system in the usb, this works properly enough because it's the same distro, otherwise you might run into breakage if the kernel config was radically different. 

I don't recommend you attempt this unless you're comfortable with GRUB, there shouldn't be any danger of losing data, but definitely danger of losing sanity

Cheers and good luck

ImanolBarba, 2024-08-16:
Also if I didn't convince you this is a bad workaround, did I mention you have to do this everytime you wanna boot the USB? The only chance of automating this is to write a `loader.conf` entry that would have to be manually updated everytime you update the kernel on the asahi install on disk.

danacr, 2024-08-17:
thanks @ImanolBarba ! I needed to boot from USB to encrypt the lvm, but even with your suggestion it hangs on boot :(

ImanolBarba, 2024-08-17:
Sorry to hear that, hopefully the devs will find the issue soon

stapelberg, 2024-08-22:
No, but the last time I booted a live Asahi was about a year ago, almost to the day :) 

(Nowadays, I just boot the installed version, haven’t had a need to boot a live version.)