leifliddy/asahi-fedora-usb #35: Error with initramfs

CL107, 2024-08-23:
When attempting to boot the usb - i get to the grub selection screen - i am presented with this error:
```
error: ../../grub-core/fs/fshelp.c:257:file */initramfs-6.10.5-400.asahi.fc40.aarch54+15k.img' not found.
```

This is then followed by a small text output and then an automatic reboot into the default os.

[image]

Unsure how to fix this error, as i just ran the script successfully and did nothing else with the usb..

mschabhuettl, 2025-09-03:
> > `dracut regenerate-all -p --force`
> > 
> > In ./build.sh chroot did the trick for me
> 
> Where in build.sh did you add that? I'm getting the same issue

Do

```
./build.sh mount
./build.sh chroot
```

Now you've change-rooted to the USB.

Just to be sure... Update root (on USB)

```
dnf clean all
dnf makecache --refresh
dnf upgrade --refresh --assumeyes
```
And then update dracula
```
dracut regenerate-all -p --force
```

Now exit USB environment

```
exit
```

And unmount USB

```
./build.sh umount
```

tgmachina, 2025-10-01:
I also ran into this same issue and @mschabhuettl's comments lead me in the right direction.

My error message also complained about the `initramfs` image not being found:

```
error: ../../grub-core/fs/fshelp.c:257:file `/initramfs-6.16.8-400.asahi.fc42.aarch64+16k.img not found`
```

Running:

```
# NOTE: it should be --regenerate-all (a flag) not regenerate-all
dracut -v --regenerate-all -p --force
```

I saw the image gets put into `/boot`

`fshelp.c` seemed to be looking in the root folder and not `/boot`. After copying the image to `/`, I was able to successfully boot from usb.