leifliddy/asahi-fedora-usb #36: Boot stuck when initializing sdhci-pci

pachulo, 2024-10-22:
Hi! When trying to boot from the usb key the process stops after some seconds, and the last lines are `sdhci-pci` related:
[image]

Then after 20 seconds or so, the system makes an hard-reset.

After reading [this tip](https://www.reddit.com/r/AsahiLinux/comments/1g9izd4/comment/lt767ep/) from @marcan, I've tried editing the grub entry to append the `module_blacklist=sdhci_pci` to the linux line and then the USB did boot!

How could this be solved for good?

Thanks!

pachulo, 2024-11-21:
OK, as pointed out by @marcan in Reddit, the workaround to fix the USB drive, so it can boot, goes like this:
```
$ ./build.sh chroot
$ dnf install fedora-asahi-remix-scripts
$ systemd-hwdb update
$ dracut --regenerate-all --force
```

mschabhuettl, 2025-10-04:
Please merge as this would also fix Issue #35

PR #38 (2025-10-04): fix sdhci-pci bootloop
With this changes, the created USB worked for me, but I'm not really sure it it's the correct way to implement it.

I've also seen this errors when creating the USB drive:

```
### Fix sdhci_pci problem
dracut[E]: No '/dev/log' or 'logger' included for syslog logging
dracut[E]: Module 'systemd-pcrphase' depends on 'tpm2-tss', which can't be installed
dracut[E]: Module 'systemd-pcrphase' depends on 'tpm2-tss', which can't be installed
```

Fixes #36
commit 0d79d753 (2025-10-04): fix: sdhci-pci bootloop

Fixes #36
closed 2025-10-04 by leifliddy