leifliddy/asahi-fedora-usb #39: Support for Asahi Fedora Remix 42

mschabhuettl, 2025-05-04:
Hi Leif,

first of all, thanks a lot for this great tool!

As you probably know, [Asahi Fedora Remix 42](https://fedoraproject.org/wiki/Changes/AsahiRemix42) has been released recently.  
Would it be possible to update this project to support F42, or check if everything still works as expected?

According to the README, this is currently based on `mkosi v24`, which matches the version in F41 – but I assume F42 might require a few adjustments.

Thanks again for your work on this project!

Best regards,  
Matthias

leifliddy, 2025-05-04:
Ok, I just updated the project to f42. Can you please let me know if everything is working for you?
I'm testing something right now that I'll add in the next day or so. From the u-boot prompt you should be able to type the following to boot the usb drive. 
usbboot loadAddr 0:0 
Let me know if that works for you...
Thanks!

leifliddy, 2025-05-04:
I forgot to mention that this project  now nonsupports mkosi version 23 or less. I wasn't able to get mkosi working with versions > 23.

mschabhuettl, 2025-05-04:
Thanks for the quick update!

I haven’t received my M2 yet – I’m just preparing things ahead of time.  
Once it arrives, I’ll give it a try and report back.

Appreciate your help!

mschabhuettl, 2025-05-17:
I'm not able to select the USB at bootmenu

mschabhuettl, 2025-05-17:
Neither usb loadAddr 0:0 works

mschabhuettl, 2025-05-18:
Dear God... Did not RTFM... I did not set mkosi path correctly... usbboot loadAddr 0:0 works like a charm

mschabhuettl, 2025-05-19:
I am able to boot the USB but boot does not complete. I'm getting a xhci-hcd host system error. Any hints?