leifliddy/asahi-fedora-usb #40: Only mkosi version 23 and below are supported please install a compatible version to continue

vedmant, 2025-08-30:
Probably needs an update, following instructions didn't work:
Only mkosi version 23 and below are supported
please install a compatible version to continue

mschabhuettl, 2025-08-30:
I don't see what you mean with "following instructions"?

Try https://github.com/leifliddy/asahi-fedora-usb/issues/39#issuecomment-2848893676

vedmant, 2025-08-30:
@mschabhuettl I mean instructions from the readme it results in the error:

```
Only mkosi version 23 and below are supported
please install a compatible version to continue
```

> I forgot to mention that this project now now supports mkosi version 23 or less. I wasn't able to get mkosi working with versions > 23.

Yeah, but I could not install mkosi < 23. I'm not a Linux power user, dnf didn't show any previous versions for mkosi.

Maybe readme should be updated of how to install proper compatible versions of dependencies.

leifliddy, 2025-08-30:
I put a copy of mkosi v22 here:  (if you're using fedora)
https://www.leifliddy.com/mkosi-rpm/

**you can install it with:
rpm -ivh https://www.leifliddy.com/mkosi-rpm/mkosi-22-1.fc42.noarch.rpm

 you'll need to add an `exclude` statement to prevent that package from being updated

ie in `/etc/yum.repos.d/fedora.repo` add the `exclude=mkosi` statement

```
[fedora]
name=Fedora $releasever - $basearch
#baseurl=http://download.example/pub/fedora/linux/releases/$releasever/Everything/$basearch/os/
metalink=https://mirrors.fedoraproject.org/metalink?repo=fedora-$releasever&arch=$basearch
enabled=1
countme=1
metadata_expire=7d
repo_gpgcheck=0
type=rpm
exclude=mkosi
gpgcheck=1
...
```